thonny/thonny #1408: AttributeError: 'NoneType' object has no attribute 'lineno'

ysarasty, 2020-11-09:
common.py", line 113, in range_contains_smaller
    this_start = (one.lineno, one.col_offset)

aivarannamaa, 2020-11-10:
Could you please post the whole stacktrace and also the script you were debugging?

ysarasty, 2020-11-10:
hand_arr = [1,2,3,5,6,9]
wildcard_rank =12
hand_arr = arrangement.get_arrangement(hand_arr, wildcard_rank)
Traceback (most recent call last):
  File "C:\Users\yuri_\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 1744, in _trace
    return self._trace_and_catch(frame, event, arg)
  File "C:\Users\yuri_\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 1789, in _trace_and_catch
    self._handle_progress_event(frame.f_back, event, marker_function_args, node)
  File "C:\Users\yuri_\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 1847, in _handle_progress_event
    self._save_current_state(frame, event, args, node)
  File "C:\Users\yuri_\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 1891, in _save_current_state
    or not range_contains_smaller_or_equal(prev_root_expression, focus)
  File "C:\Users\yuri_\AppData\Local\Programs\Thonny\lib\site-packages\thonny\common.py", line 129, in range_contains_smaller_or_equal
    return range_contains_smaller(one, other) or one == other
  File "C:\Users\yuri_\AppData\Local\Programs\Thonny\lib\site-packages\thonny\common.py", line 113, in range_contains_smaller
    this_start = (one.lineno, one.col_offset)
AttributeError: 'NoneType' object has no attribute 'lineno'

aivarannamaa, 2020-11-11:
Thank you for the stacktrace, but I still can't reproduce the problem. Please send the whole script that caused it! If you sent it as attachment, then it doesn't work -- GitHub does not read attachments sent by e-mail. Please post at GitHub instead.

closed 2021-01-16 by aivarannamaa