thonny/thonny #1410: Micropython: another autocomplete issue (editor view)

adzierzanowski, 2020-11-10:
Autocompletions work fine in the terminal view but with micropython files there's no autocompletion at all for user-defined modules. Consider:

```python
# foo.py
def foo():
    print('bar')
```

```python
# bar.py
import foo

foo.#<TAB> hit here, nothing happens
```

Would it take much to implement autocompletions in the editor view? I suppose it would require loading the imported modules dynamically and would raise the issue of cluttering the board's communication port if there were many large modules used.

aivarannamaa, 2020-11-10:
Do you edit your scripts on the device, or on the local machine? In the latter case it may require some tweaking in Thonny, but I don't have a good solution for the former.

adzierzanowski, 2020-11-10:
On the device, unfortunately.

aivarannamaa, 2020-11-10:
I could maintain a local shadow copy of each file from the device, if it has been opened in Thonny (in this case no extra data needs to be exchanged).

I don't have time to squeeze this into version 3.3, though.

aivarannamaa, 2023-01-26:
Should be working in 4.0

closed 2023-01-26 by aivarannamaa