thonny/thonny #1416: Suggestion: remove support for Friendly-traceback

aroberge, 2020-11-14:
Suggestion: remove Thonny's built-in support for Friendly-traceback until version 1.0 is available.  I can do this in a pull request if desired.

Friendly-traceback (currently at version 0.1.11) can now be run from any editor, using a secondary file as a launcher. 
Here's a screen capture using Thonny:

[image]

As a result, built-in support for Thonny, while slightly more convenient (in theory) is no longer needed.  Removing code related to Friendly-traceback from within Thonny would ensure that no breakage would occur when changes to Friendly-traceback are made.

Once the API of Friendly-traceback is "frozen", the idea of adding direct support for Friendly-traceback within Thonny could be revisited.

aivarannamaa, 2020-11-14:
Alright, I don't mind. I'll remove it before releasing final 3.3.0

aroberge, 2020-11-14:
And I want to really thank you for the early support you gave to Friendly-traceback. I have been adding some features found in Thonny's assistant, mostly inspired by some comments you made a while ago on the design of Friendly-traceback.

aivarannamaa, 2020-11-14:
I'm glad my comments have been useful! I wish you luck!

closed 2020-11-15 by aivarannamaa