thonny/thonny #1422: Autocomplete OSX

juanfal, 2020-11-15:
I never used autocomplete in Thonny before.  Now that it works better and better, I am editing everything with it, but I never have seen autocomplete working on OSX and with last 3.3 it still doesn't.  I press TAB after **for** and a menu appear with **for**/**format**, but **^S** seems that does nothing at all.  I'd like it worked with a simple repetition of another TAB, like in Atom/Sublime…

Thanks

aivarannamaa, 2020-11-16:
Thank you for the report!

I can't reproduce the problem with ^S in my Catalina Mac.
Regarding several TAB-s -- I plan to enhance code completion in Thonny 4.0, so I'm postponing it for now.

juanfal, 2020-11-16:
Thank you Aivar.  I’ve reviewed some of the Spanish sentences (a few from myself!) after reading them on 3.3.  Great system!!

With respect to the autocompletion, I’ve seen variables autocomplete perfectly well, just pressing TAB.  But, as I said, for keywords, like for, the menu with the opctions (for/format in that case) pops up, but you can’t do anything there, since ^S doesn’t work either from the very keyboard or from the Edit→Autocompletar.  It blinks when you type ^S but doesn’t work in any way.

PD.: Thonny is becoming a great environment.  I teach C-C++ with other group of first year students, and I miss having this Thonny for C, too!  
 
- Juan Falgueras

aivarannamaa, 2020-11-16:
@juanfal , regarding C/C++, do you know these tools:

* https://sites.google.com/site/virtualcide/
* https://seec-team.github.io/seec/seec-view.html

? 

I've not used them myself, but they look interesting.

juanfal, 2020-11-16:
Hi Aivar

	They look and work really well, at least the first one, and the second says the same.  But it seems both only let compile C and have no possibility of compiling any .cpp, so that’s torn it!

Anyway, I’m learning I know nothing.

Thank you Aivar
- Juan F.

aivarannamaa, 2022-04-01:
Closing as completions are now better in version 4.0.0b1

closed 2022-04-01 by aivarannamaa