thonny/thonny #1423: CTRL+F  not working sometimes

krupis, 2020-11-16:
I am noticing this strange behaviour in Thonny witl CTRL+F. Sometimes using this shortcut will not open the find menu. This happens about 70% of the time I use Thonny. One hour it might work fine, the next it will completely stop working. I am not sure whether I am accidentally disabling it somehow?? I really need to fix it - cant work with Thonny without ctrl+f

aivarannamaa, 2020-11-16:
Which platform and which Thonny version are you using?

When Ctrl+F doesn't do anything, can you invoke the dialog from "Edit" menu?

Can it be that you had Caps Lock on while you were pressing Ctrl+F and nothing happened? Unfortunately Thonny's UI framework is specific about this.

krupis, 2020-11-16:
Yes indeed it was caps lock! Thanks for very quick response. I was not
aware that ctrl+f does not work when caps lock is pressed

aivarannamaa, 2020-11-16:
I'll try to find a work-around for this problem under #1347

closed 2020-11-16 by aivarannamaa

pssolanki111, 2020-11-17:
@aivarannamaa Hey Aivar, Just a quick question. Based on your mention of #1347, is thonny's UI written with `tkinter`? If yes, then what makes it so incredibly better than IDLE in editing and shell scripting 'cause IDLE is written in `tk` too?

aivarannamaa, 2020-11-17:
@pssolanki111, yes, Thonny is based on Tkinter. 

I guess the IDLE developers had different goals (minimalism, probably).