thonny/thonny #1426: Thonny crashes with tkinter.TclError

vaenda, 2020-11-16:
Arch Linux, Thonny 3.3.0 installed by bash script from thonny.org without errors.
When I try to start up thonny with apps/thonny/bin/thonny I get the error:

[thonny-error.txt](https://github.com/thonny/thonny/files/5549056/thonny-error.txt)

I had the same problem with Thonny 3.2.7 and have found no information how to solve it.

I would appreciate any help. If I need to provide further information, please ask.

aivarannamaa, 2020-11-17:
Could you please give me instructions for installing similar system so that
i get same version, wm etc?

vaenda, 2020-11-18:
Thanks for your answer. 

To give instructions for installing arch linux is beyond the scope of my answer (and probably not what you meant). The List of installed packages in my system can be found here: 
[package-list.txt](https://github.com/thonny/thonny/files/5559754/package-list.txt)
or - with version numbers:
[package-list-with-version-numbers.txt](https://github.com/thonny/thonny/files/5559756/package-list-with-version-numbers.txt)
Only official repos, I purged all AUR packages from the system.
I am using sddm as display-manager together with xfce4 and xorg-server, but in gnome I get the same error message.

aivarannamaa, 2020-11-18:
Maybe you can find something when you grep `Mono:size=20:style=medium:antialias=true` through your system?

vaenda, 2020-11-18:
That seems plausible, but a grep -R didn't find anything, so I will just have to live with it (or reinstall). In my debian install on school PCs it works and since I primarily want to use it with students in school that is sufficient. My home PC is my problem, not yours, so thanks for your work! I will find out what the problem is eventually (or not). ;-)

vaenda, 2020-11-19:
It turns out, that I had modified my .Xresources file and added some lines to ensure antialiasing by all programs. 
Additionally I must have been too stupid to use grep correctly, otherwise I would have found the error earlier. 
I apologize for the waste of time. Thanks again for your help.

aivarannamaa, 2020-11-19:
No problem! I'm glad you found it.

closed 2021-01-16 by aivarannamaa