thonny/thonny #1434: Printing has different theme as in thonny-editor

speedy-10, 2020-11-18:
I've chosen in Options > "Syntax Theme" > IDLE Classic.
If I print the quellcode via Firefox, the code gets the default syntax-highlighting of thonny, but not the chosen one.

thonny 3.3.0  (I discovered that in earlier Versions too).

aivarannamaa, 2020-11-18:
For printing Thonny uses fixed colors.

It can't simply use selected syntax theme, because people who use dark themes most likely don't want to print light letters on dark background (or without background).

In principle I could make it configurable, whether to use fixed theme or current theme, but IMO this is not important enough to warrant a new item in the configuration dialog (I don't want to make it too crowded).

I guess I should add a flag to each syntax theme, which shows whether it is suitable for printing or not and use the fixed theme for printing only if it is not.

speedy-10, 2020-11-18:
Hi, yes a flag would be nice and fits for my belongings. Thanks for your work.