thonny/thonny #1438: Please add keyboard shortcut to menu: Run->Disconnect

IhorNehrutsa, 2020-11-20:
Possibly Shift + F2.
Thanks

aivarannamaa, 2020-12-01:
Sorry for late reply!

I assume that you want the shortcut, because you need to disconnect often. Is it because you want to connect to same device from different program?

IhorNehrutsa, 2020-12-02:
Yes, You are right. I am developing C modules for ESP32 MicroPython 
https://github.com/micropython/micropython/pulls/IhorNehrutsa
and I need to switch to esptool.py to write the firmware to the chip 
and then switch back to Thonny to run Python code.
Thanks.

aivarannamaa, 2020-12-02:
I see. I think you are in a rather unusual situation, compared to other Thonny users. Therefore I'd rather not use up a shortcut key for this command, but you can add it yourself: https://github.com/thonny/thonny/wiki/Custom-shortcuts

IhorNehrutsa, 2020-12-02:
I added 
```
[shortcuts]
disconnect = <Shift-F2>
```
to c:\Users\ < user >\AppData\Roaming\Thonny\configuration.ini 
and I have the shortcut and it works as expected
[image]
Am I happy?
Not fully.
I need to do the same in 3 other places: in VirtualBox, in VMware Player, and on my notebook :(

aivarannamaa, 2020-12-02:
Well, the choice is yours...

closed 2021-01-16 by aivarannamaa