thonny/thonny #1442: ManagementError on reset when main.py is present on microbit

r0the, 2020-11-22:
In Thonny 3.3.0, when the reset button of a microbit device is pressed and a main.py file
is present on the microbit device, a ManagementError occurs:

```
>>> THONNY FAILED TO EXECUTE COMMAND get_globals

SCRIPT:
__thonny_helper.print_mgmt_value({name : __thonny_helper.repr(value) for (name, value) in globals().items() if not name.startswith('__')})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '__thonny_helper' is not defined
```

The expected behaviour is that the main.py file is executed and no error occurs. In Thonny 3.2.7, this works correctly.

aivarannamaa, 2020-11-22:
Thank you for the report!

I'll try to fix this in 3.3.1

closed 2020-11-26 by aivarannamaa

PeterNoelEvans, 2020-11-29:
I have this problem too. I am running main.py and ssd1306.py on ESP32 WROOM. I can load the files fine and run them with F5 with no errors, however, when I reboot the ESP32, it gives me this error:

To get out of it, I have to STOP - wait for the REPL (which now requires the second try after 1 second, and then the REPL comes back. I can edit the code, save, and RUN with F5 and it works fine.

`>>> THONNY FAILED TO EXECUTE COMMAND get_globals

SCRIPT:
__thonny_helper.print_mgmt_value({name : __thonny_helper.repr(value) for (name, value) in globals().items() if not name.startswith('__')})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '__thonny_helper' isn't defined`

PeterNoelEvans, 2020-11-29:
I installed Thonny 3.2.7 and the problem went away.

aivarannamaa, 2020-11-29:
I confirmed that the fix works for ESP32 as well (Sparkfun ESP32 Thing)

toLucky81, 2021-03-28:
Hi, Im having the same problem.

Running:
Thornny 3.3.6
Ubuntu 20.10 64bit
Python 3.7.9 64bit
Tk 8.6.10

CODE:
 (from github example https://github.com/raspberrypi/pico-micropython-examples/blob/master/multicore/multicore.py#L1-L12 )

import time, _thread, machine

def task(n, delay):
    led = machine.Pin(25, machine.Pin.OUT)
    for i in range(n):
        led.high()
        time.sleep(delay)
        led.low()
        time.sleep(delay)
    print('done')

_thread.start_new_thread(task, (10, 0.5))

OUTPUT
THONNY FAILED TO EXECUTE COMMAND get_globals

SCRIPT:
__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '__thonny_helper' isn't defined

%Run -c $EDITOR_CONTENT
THONNY FAILED TO EXECUTE COMMAND Run

SCRIPT:

class __thonny_helper:
    try:
        import uos as os
    except ImportError:
        import os
    import sys

    # for object inspector
    inspector_values = dict()
    last_repl_values = []
    @classmethod
    def print_repl_value(cls, obj):
        if obj is not None:
            cls.last_repl_values.append(obj)
            cls.last_repl_values = cls.last_repl_values[-5:]
            print('[object_link_for_thonny=%d]' % id(obj), repr(obj), '', sep='')

    @staticmethod
    def print_mgmt_value(obj):
        print('<thonny>', repr(obj), '</thonny>', sep='', end='')

    @staticmethod
    def repr(obj):
        try:
            s = repr(obj)
            if len(s) > 50:
                s = s[:50] + "..."
            return s
        except Exception as e:
            return "<could not serialize: " + str(e) + ">"

    @classmethod
    def listdir(cls, x):
        if hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @classmethod
    def getcwd(cls):
        if hasattr(cls, "getcwd"):
            return cls.os.getcwd()
        else:
            # micro:bit
            return ""

    @classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 6
SyntaxError: invalid syntax

ManagementError
>>> THONNY FAILED TO EXECUTE COMMAND get_globals

SCRIPT:
__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '__thonny_helper' isn't defined

done

aivarannamaa, 2021-03-28:
@toLucky81, thank you for the report, but your case is actually different. See #1586