thonny/thonny #1443: interpreter tab page disappeared on some Windows PC

codemee, 2020-11-23:
On some Windows PCs that workd fine with Thonny 3.2.7, I installed Thonny 3.3.0 but met the strange situation after selecting MicroPython(ESP8266) in the interpreter tab of options dialog, if I plug in a D1 mini board and reopen the options dialog, the Interpreter tab disappeared.
[image]
Then if i press Stop to reconnect to the D1 mini board Thonny would report error.
[image]
If I restart Thonny, a similiar error reported
[image]
If i unplug the board, the interpreter tab would come back.
I've test 5 pcs in my office, 2 of them got the strange things showed above but I don't know why?

aivarannamaa, 2020-11-23:
Thank you for the report! I could reproduce this.

```
Traceback (most recent call last):
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\backend_config_page.py", line 122, in _backend_changed
    page = self._get_conf_page(backend_desc)
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\backend_config_page.py", line 141, in _get_conf_page
    self._conf_pages[backend_desc] = cp_constructor(self.labelframe)
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\esp\__init__.py", line 98, in __init__
    super().__init__(master, "esp8266", "0x0")
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\esp\__init__.py", line 82, in __init__
    super().__init__(master)
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 409, in __init__
    for p in list_serial_ports()
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 873, in list_serial_ports
    return list(serial.tools.list_ports.comports())
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\serial\tools\list_ports_windows.py", line 412, in comports
    return list(iterate_comports())
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\serial\tools\list_ports_windows.py", line 334, in iterate_comports
    info.serial_number = get_parent_serial_number(devinfo.DevInst, info.vid, info.pid)
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\serial\tools\list_ports_windows.py", line 197, in get_parent_serial_number
    vid = int(m.group(1), 16)
AttributeError: 'NoneType' object has no attribute 'group'
```

I suspect it has to do with updated Pyserial version. I'll see what can be done. I the meanwhile you could try downgrading your pyserial version:

* Select "The same interpreter which runs Thonny" backend
* Select "Tools => Open system shell"
* enter `pip install pyserial==3.4`

aivarannamaa, 2020-11-23:
It looks like Pyserial has released final version of 3.5 with this bug fixed, so please change `3.4` to `3.5` in my last post.

closed 2020-11-23 by aivarannamaa

codemee, 2020-11-23:
I'll try this tomorrow,thanks.