thonny/thonny #1444: allow editing of "other" text based files

aoloe, 2020-11-25:
thonny should allow the editing of textual files that are not "pure" text.

the most important being markdown files.

other important candidates:

- json
- yaml
- csv
- rst (restructuredText)

also, i wonder if by default thonny should show files in "Supported formats" and allow the user to restrict the choice to "Python files" (currently the default being "Python files" and the user being allowed to chose "Text files" or "All files")

syntax highlight and other goodies would nice, but being able to open those files would already be wonderful (and probably necessary for many people!)

aoloe, 2020-11-25:
btw, if you agree that this goes in the right direction, i can try to make a patch...

aivarannamaa, 2020-11-25:
Which Thonny version are you using? Since 3.3 Thonny can show all kinds of text files (without syntax highlighting).

aoloe, 2020-11-26:
hi @aivarannamaa 

perfect then!

i'm still on 3.2.7 as distributed by debian testing.
but i will be able to test it soon on my friend's computer who is _really_ struggling with editing files on windows : - )

closed 2020-11-26 by aoloe

aivarannamaa, 2020-11-26:
> i'm still on 3.2.7 as distributed by debian testing.

I'm not sure when or whether 3.3.0 will make it into Debian, as [one of its dependencies is having problems](https://tracker.debian.org/pkg/thonny).

aoloe, 2020-11-27:
ok, i'm not an heavy user of thonny personally... and i can always install it through pip : - )

but i will need to take care that i'm not on the latest release before making further proposals.

i had a look at thonny 3.3.0 and now i can indeed open all kind of text files.
i have one feedback: i would remove the explicit "txt" file type and replace it by "All supported formats" (with a known list of good extensions for text based files (md, json, yaml, csv ...) and files (`.gitignore`...).
If it's not too hard to do...

aivarannamaa, 2020-11-27:
> i would remove the explicit "txt" file type and replace it by "All supported formats" (with a known list of good extensions for text based files (md, json, yaml, csv ...) and files (.gitignore...)

I'm afraid the solution would be incomplete for someone. What do you think of removing this option altogether and leaving only "Python files" and "all files" ?

aoloe, 2020-11-30:
you're probably right.

on the one side, i'd like to to have all relevant files to be shown by default, on the other side showing all files might be distracting.

there is only one thing i'm sure of: around me (almost) nobody needs to open `*.txt` files and having an entry for this type of files feels wrong (when all interesting flies are in "all")

if you don't want to maintain a list of supported extensions (which i fully understand), i think that my favorit would be "py" + "all" and retaining the last choice made by the user.

aivarannamaa, 2020-12-14:
In Thonny 3.3.1 it's now two choices: "Python files" and "All files".

There is no straightforward way to retaining the last choice made by the user, as the dialog only gives me the path, nothing else.