thonny/thonny #1445: pixed icon "stop" with 4k

speedy-10, 2020-11-26:
Hi, I discovered that the stop icon is unsharp with 4k monitor: UI scaling 4.0.
See picture

Thonny 3.3.0  (ubuntu 20.04).

Regards

aivarannamaa, 2020-11-26:
Thanks!

I'll keep it in the TODO list, but I don't know when will I find time for this.

If you want to get it faster, then you could create a 32x32 icon yourself.

speedy-10, 2020-11-26:
No problem, I only want to help to improve this nice editor.

speedy-10, 2022-01-31:
hi, will this issue be fixed with Thonny 4.0 :-) ?

speedy-10, 2022-01-31:
this picture is offered with:  CC BY-NC 4.0 Licence
https://pngset.com/download-free-png-izmoz
So I reduced the size to 32x32 pixel, see attachement. 
Regards

[image]

aivarannamaa, 2022-01-31:
Thanks! This looks really nice! It will be in 4.0

closed 2022-01-31 by aivarannamaa