thonny/thonny #1448: Cannot select a virtual enviroment in dot directory

m4r35n357, 2020-11-28:
Dot folders are invisible to the file selector.  The virtualenvwrapper package on Rapberry Pi OS installs environments to ~/.virtualenvs, so they become invisible to the Thonny file selector.

I had to make a **symlink** to access mine, so this is not fatal, but is inconvenient and potentially frustrating to users attempting to use virtual environments for the first time.

aivarannamaa, 2020-11-29:
Try right-clicking on the file selection dialog -- it should give you an option for showing hidden files.

m4r35n357, 2020-11-29:
You are absolutely correct,  I am not used to the Pi file selector!  Sorry to trouble you with this.  BTW Thonny has taken over from my attempts to understand Spyder on the Pi ;)

closed 2020-11-29 by aivarannamaa