thonny/thonny #1452: I broke Thonny

pyobox, 2020-11-29:
So I beleieve I accidentally put thonny in expert mode, therefore the print appeared too small, so I tried to adjust the UI size and I accidentally chose the wayyy wrong size.  Now the print is so big only a couple of words fit on screen.  I can't use the options, tools, anything because literally none of it fits on display.  I tried uninstalling and reinstalling thonny and I've even tried switching to a bigger display.  Using any built in options is out of the question because I either can't see my selection or simply the mouse will not let me select anything off screen, which again is all of it.  So how do I fix thonny without having to completely factory reset my pi

aivarannamaa, 2020-11-29:
The easiest solution:

* close Thonny
* go to `~/.config/Thonny`
* delete `configuration.ini`
* start Thonny again

pyobox, 2020-11-29:
Worked thank you so much for the fast response!

closed 2020-11-29 by pyobox