thonny/thonny #1454: Literal type autocompletion?

illume, 2020-11-29:
[Literal types](https://www.python.org/dev/peps/pep-0586/#abstract) let you specify valid inputs, which potentially provides an elegant interface...

```python
SystemCursors = Literal["arrow", "ibeam", "wait", "crosshair", "waitarrow",
    "size_nwse", "size_we", "size_ns", "size", "no", "hand"]

def set_cursor(cursor: SystemCursors) -> None:
    pass

set_cursor("wait") #autocomplete on the literals would be nice here.
```

I'd hope that IDEs like Thonny could eventually support Literal types as a command line autocomplete. I'm not sure of any that do so far however.

Related pygame issue: https://github.com/pygame/pygame/pull/2331

aivarannamaa, 2020-11-29:
Thonny uses [jedi](https://github.com/davidhalter/jedi) library for auto-completion, so it's better to report this proposal there.

illume, 2020-11-29:
Thanks @aivarannamaa

closed 2020-11-29 by illume