thonny/thonny #1456: About adding modules

asp65, 2020-12-01:
i don't know why i've added this module, but it doesn't work.
[image]
[image]

aivarannamaa, 2020-12-01:
In the screenshot you have installed it for Python 3 in your computer, but according to the package description it is meant for MicroPython, ie. it should be installed on the device.

Which MicroPython device are you using?

asp65, 2020-12-01:
> In the screenshot you have installed it for Python 3 in your computer, but according to the package description it is meant for MicroPython, ie. it should be installed on the device.
> 
> Which MicroPython device are you using?

i didn't install it on my device. now,it is working. thank you guys for helping! my mircopython is esp32, other device isn't being sold in china.

closed 2020-12-01 by aivarannamaa