thonny/thonny #1457: Feature Request: add an option to disable hash randomization

raenye, 2020-12-01:
Python is using random salt for `hash` of `str` and `bytes` objects since 3.3. It is possible to disable this by setting the environment variable [PYTHONHASHSEED](https://docs.python.org/3/using/cmdline.html#envvar-PYTHONHASHSEED) to 0, but this has to be done **before** the interpreter starts.

See, e.g., [a related stackoverflow question](https://stackoverflow.com/questions/61726053/how-to-set-pythonhashseed-in-python-file).

Obviously it's possible to set environment variables globally, but I thought it'd be easier if there was a checkbox in the settings that allows this to happen only in Thonny.

Thanks in advance :smile:

aivarannamaa, 2020-12-01:
Thank you for the suggestion!

I just added (in version 3.3.0) possibility to specify environment variables for Thonny's GUI process (which will be also inherited by the backend process running user code). Set the variable in "Tools => Options => General" and restart Thonny. According to my tests, it seems to work.

I hope this will be good enough for you, because I'd prefer not to add another checkbox for this.

raenye, 2020-12-02:
Thanks @aivarannamaa, I just updated to 3.3.0.

After adding `PYTHONHASHSEED=0` there, `hash` is deterministic once again.

Works for me, so I'm closing the request.

closed 2020-12-02 by raenye