thonny/thonny #1460: Make indent size configurable

aivarannamaa, 2020-12-03:


zino113, 2022-11-08:
An option entry in 'configuration.ini' like
auto-indent=2 
would be nice to have ...
(maybe less work than adding it to >Editor >Options).
---
I'm a Python newbie and like your uncluttered IDE -- ideal for smaller projects and experiments. Thank you.
---

canepan, 2023-12-17:
As an additional point to support this, [Pimoroni enviro](https://github.com/pimoroni/enviro/tree/main/enviro) files use 2 spaces indentation, and it looks like one of main Thonny use cases is MicroPython on Pico devices

resistantJs, 2024-02-09:
Is there any possibility that we could change this on the user side with some tweaking in the "possible configuration file" instead of waiting for a later update?

aivarannamaa, 2024-02-25:
> Is there any possibility that we could change this on the user side with some tweaking in the "possible configuration file" instead of waiting for a later update?

Unfortunately not and it's not easy to pinpoint single place in code for changing this either.

Anyway, I implemented it now and the configuration option will be present in Thonny 5.0.

closed 2024-02-25 by aivarannamaa