thonny/thonny #1461: ERROR   thonny.plugins.micropython.backend: Crash in backend

profra, 2020-12-04:
Working with "ATOM LITE" from M5STACK (ESP32-PICO) on OS Linux Mint 20 Cinnamon I received the following report...
...repeatedly... Thonny 3.3.0 installed
----------------------------------------------------------------------------------------------------------------------------------------------------------------

MicroPython v1.13-221-gc8b055717 on 2020-12-03; ESP32 module with ESP32
Type "help()" for more information.
ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 44, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "/home/fp/apps/thonny/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 130, in __init__
    self._prepare(clean)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 149, in _prepare
    self._prepare_helpers()
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 196, in _prepare_helpers
    self._execute_without_output(script)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 544, in _execute_without_output
    out, err = self._execute(script, capture_output=True)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 518, in _execute
    self._execute_with_consumer(script, consume_output)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 548, in _execute_with_consumer
    self._submit_code(script)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 524, in _submit_code
    self._connection.read_all_expected(expected_echo, timeout=WAIT_OR_CRASH_TIMEOUT)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 115, in read_all_expected
    actual = self.read(len(expected), timeout=timeout)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 49, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'     return s\n        except Exception as e:\n            return "<could no serialize: " + str(e) + ">"\n\n    @classmethod\n    def listdir(cls, x):\n        if hasattr(cls.os, "listdir"):\n            return cls.os.listdir(x)\n        else:\n            return')
Traceback (most recent call last):
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 44, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "/home/fp/apps/thonny/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 130, in __init__
    self._prepare(clean)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 149, in _prepare
    self._prepare_helpers()
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 196, in _prepare_helpers
    self._execute_without_output(script)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 544, in _execute_without_output
    out, err = self._execute(script, capture_output=True)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 518, in _execute
    self._execute_with_consumer(script, consume_output)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 548, in _execute_with_consumer
    self._submit_code(script)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 524, in _submit_code
    self._connection.read_all_expected(expected_echo, timeout=WAIT_OR_CRASH_TIMEOUT)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 115, in read_all_expected
    actual = self.read(len(expected), timeout=timeout)
  File "/home/fp/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/connection.py", line 49, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'     return s\n        except Exception as e:\n            return "<could no serialize: " + str(e) + ">"\n\n    @classmethod\n    def listdir(cls, x):\n        if hasattr(cls.os, "listdir"):\n            return cls.os.listdir(x)\n        else:\n            return')

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2020-12-04:
Thank you for the report!

It looks like communication issue with the device -- Thonny is sending a command to Atom and it expects to read identical echo, but in the the echo (at least) one character is missing.

Did you try unplugging and re-plugging the device?

Do you have stock firmware on it? UIFlow?

profra, 2020-12-04:
Hi,
as you can see from the first line of the report, MicroPython is installed on ATOM LITE, not UIFlow.
FYI CircuitPython on Seeeduino XIAO (SAMD21) seems to work well just like MicropPython on WEACTF411CE (STM32, F4).
But I need so much functionality on esp32... I believe I'm not alone :-))

aivarannamaa, 2020-12-04:
I have several ESP32 devices, which work fine in Thonny 3.3.0, but unfortunately no ESP32-Picos. M5stack will have sale in 3 days, I'll order an ATOM LITE then.

profra, 2020-12-04:
You motivated me to flash MP to other modules (from Adafruit, THINGPULSE, and others), you're right, they work all well.
FYI also ATOM MATRIX behaves the same as LITE, both with the same ESP32-PICO.
!!! FYI I also did the test with the original ESPRESSIF ESP32-PICO-KIT_V4 board and Thonny behaves normally!!!
**It all follows that the problem may not be in esp32-pico but in ATOM LITE and ATOM MATRIX...**

aivarannamaa, 2020-12-04:
If you are using ESP32 back-end, then there is something to try with Thonny 3.3:

* Select "Tools =>  Open Thonny data folder"
* close Thonny
* open configuration.ini
* Add following to `[ESP32]` section: `write_block_size = 64` and save
* Reopen Thonny

profra, 2020-12-04:
I tried it but it didn't help.

jessevl, 2020-12-16:
Same here on M5StickC and M5StickC Plus. 3.2.7 works fine, but exactly the same error messages with 3.3.0+. I'm thinking issue #1480 may also be the same.

profra, 2020-12-16:
FYI... nothing has changed even in version 3.3.1

aivarannamaa, 2020-12-16:
Has any of you experienced the same problem with the big M5Stack?

profra, 2020-12-19:
> Has any of you experienced the same problem with the big M5Stack?

I tried M5Stack Core Basic (black) and I can confirm that it works with Thonny 3.3.1 without any problems.

ukdavewood, 2020-12-21:
Same problem for me using an m5stickC with uflow 1.7.0 firmware installed.   Installed Thonny 3.2.7 and able to connect ok.

aivarannamaa, 2021-01-05:
Should be fixed under #1516, will be released in Thonny 3.3.2 (hopefully in a week). Please reopen, if you are still having same problem in Thonny 3.3.2

closed 2021-01-05 by aivarannamaa