thonny/thonny #1462: ESP32: Run-Disconnect breaks the microcontroller program

IhorNehrutsa, 2020-12-04:
Latest Thonny, latest stable ESP32 port of MicroPython.
I connect ESP32 to my WIFI router.
When I try to disconnect, I suppose that the ESP32 still working, but it is not.
WebREPL lost the connection (reconnect not possible).
[image]

There is no that effect in [EsPy IDE](https://github.com/jungervin/EsPy).
WebREPL connected to the ESP also works synchronously to the terminal inside IDE.
[image]

aivarannamaa, 2020-12-04:
Thank you for the report! I wish all reports were so clearly presented!

Currently the "Disconnect" simply kills the back-end process, which is talking to the device.

I have noticed that when I'm opening connection to some of my devices with PySerial, the device resets. It looks like the same can happen on disconnect. I suspect that there is a difference in how PySerial connects to the device and how EsPy does.

I couldn't reproduce this behavior in Ubuntu with Sparkfun ESP32 thing. Will try Windows later.

aivarannamaa, 2020-12-04:
Can you test whether the device reboots on disconnect from Thonny or only the webepl service disappears?
When you create a variable in REPL, disconnect, exit Thonny and start Thonny again, is the variable still present?

Note that you can't use Stop/Restart in this test because it does soft-reboot by design.

IhorNehrutsa, 2020-12-04:
ESP32 flash is rewritten.
boot.py is default:
```
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
#import webrepl
#webrepl.start()
```
Open Thonny 1
[image]
Close Thonny
Open Thonny 2
[image]
Close Thonny
Open Thonny 3
[image]

ESP reboots are happening, sometimes it visible in Shell, sometimes not.

IhorNehrutsa, 2020-12-05:
```
[ESP32]
dtr = False
rts = False
```
It seems that more convenient and right name of section in ini file is:
[USB_UART]
not
[ESP32]
because CP2102 used in Arduino/AVR, STM32 boards also.

And thanks for your work and time.

aivarannamaa, 2020-12-05:
All MicroPython settings are grouped under respective back-end, ie. when you want to set dtr/rts for ESP32, then they must be under `[ESP32]`. If you need it for other kinds of devices as well, then you also need to set the same settings under other sections.

IhorNehrutsa, 2020-12-05:
Ok, thanks for clarifying.

closed 2020-12-06 by aivarannamaa

IhorNehrutsa, 2020-12-07:
Tested in Thonny 3.3.1. All right.

slabua, 2022-01-02:
Still needed to be added in 3.3.14.