thonny/thonny #1463: ERROR: AssistantView & Internal Error Messages on Thonny 3.3.0 for Windows

HarshPatel0304, 2020-12-04:
I have been using Thonny 3.3.0 for about 2 weeks now and didn't experience any issues until yesterday (Friday, Dec 3rd, 2020) when I launched Thonny and received a error message (see below, AssistantView Error Message) and then my Assistant View stopped working even though the toggle option was set to enable it, I restarted my computer and tried launching Thonny again and got this time I didn't just have that AssistantView Error Message, I also received an Internal Error Message (see below, Internal Error Message) and after that I was able to launch Thonny but no matter where I clicked on the program, I couldn't access anything in the program.. From there I uninstalled Thonny and deleted all related files I could find under Program Files on my computer and deleted the application as well, then I went to the Thonny website and re-downloaded Thonny 3.3.0 for Windows and installed but I had the same exact issues... Then I tried doing a "Troubleshoot Compatability" Test after launching it for the second time and that seemed to fix the part where I couldn't click on anything on the program so right now I can access the main features I use such as the shell and code window etc., but I am still getting the same two error messages (about the AssistantView and Internal Error) and the AssistantView is still not working... I went under the "Help" section and clicked on "Help contents" and it gave me another Internal Error Message (see below, Internal Error Message under "Help" --> "Help contents"). What can I do to fix this?

(AssistantView Error Message)  
``` 
Traceback (most recent call last):
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1445, in _show_views
    self.show_view(view_id, False)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1622, in show_view
    view = self.get_view(view_id)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1528, in get_view
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\assistance.py", line 48, in __init__
    insertwidth=0,
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 780, in __init__
    self.text = text_class(self, **final_text_options)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\rst_utils.py", line 25, in __init__
    self.configure_tags()
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\assistance.py", line 406, in configure_tags
    super().configure_tags()
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\rst_utils.py", line 67, in configure_tags
    font=underline_font
TypeError: tag_configure() got multiple values for keyword argument 'font'

```

(Internal Error Message)
``` 
Traceback (most recent call last):
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 241, in _try_action
    action()
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 2354, in _open_views
    self.get_view("AssistantView")
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1528, in get_view
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\assistance.py", line 48, in __init__
    insertwidth=0,
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 780, in __init__
    self.text = text_class(self, **final_text_options)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\rst_utils.py", line 25, in __init__
    self.configure_tags()
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\assistance.py", line 406, in configure_tags
    super().configure_tags()
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\rst_utils.py", line 67, in configure_tags
    font=underline_font
TypeError: tag_configure() got multiple values for keyword argument 'font'

```

(Internal Error Message under "Help" --> "Help contents")
```
Traceback (most recent call last):
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1044, in dispatch_from_menu
    dispatch(None)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 986, in dispatch
    handler()
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\help\__init__.py", line 70, in open_help
    get_workbench().show_view("HelpView")
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1622, in show_view
    view = self.get_view(view_id)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1528, in get_view
    view = class_(self)  # View's master is workbench to allow making it maximized
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\help\__init__.py", line 30, in __init__
    read_only=True,
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 780, in __init__
    self.text = text_class(self, **final_text_options)
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\rst_utils.py", line 25, in __init__
    self.configure_tags()
  File "C:\Users\gotoj\AppData\Local\Programs\Thonny\lib\site-packages\thonny\rst_utils.py", line 67, in configure_tags
    font=underline_font
TypeError: tag_configure() got multiple values for keyword argument 'font'

```

Ivoz, 2020-12-05:
Have you installed any fonts recently?

Changing the fonts thonny uses *might* work around it for now.

aivarannamaa, 2020-12-05:
@HarshPatel0304 , thank you for the report!

At the moment I don't have any ideas about what may have caused it, but your Thonny's configuration file may give some hints. You can find it by selecting "Tools => Open Thonny data folder", the file's name is configuration.ini. Please post it here.

HarshPatel0304, 2020-12-05:
> Have you installed any fonts recently?
> 
> Changing the fonts thonny uses _might_ work around it for now.

No, I haven't installed any fonts for Thonny or anything in general, I did however get the Dracula Theme Plug-in for Thonny although I've been using it since the first day I got it so I don't think that's the issue. What font(s) might help?

HarshPatel0304, 2020-12-05:
> @HarshPatel0304 , thank you for the report!
> 
> At the moment I don't have any ideas about what may have caused it, but your Thonny's configuration file may give some hints. You can find it by selecting "Tools => Open Thonny data folder", the file's name is configuration.ini. Please post it here.

This is what it says:
```
[general]
configuration_creation_timestamp = 2020-11-23T08:57:33.822690
language = en_US
environment = []
single_instance = True
event_logging = False
disable_notification_sound = False
debug_mode = False
ui_mode = regular
scaling = default
font_scaling_mode = default

[view]
full_screen = False
maximize_view = False
shellview.visible = True
ui_theme = Clean Dark
syntax_theme = Dracula
editor_font_size = 13
editor_font_family = Consolas
io_font_size = 11
io_font_family = Courier New
assistantview.visible = True
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = True
outlineview.visible = False
variablesview.visible = True
show_program_arguments = False
show_plotter = False
name_highlighting = False
locals_highlighting = False
paren_highlighting = True
syntax_coloring = True
highlight_tabs = True
highlight_current_line = False
show_line_numbers = True
recommended_line_length = 0

[run]
backend_name = SameAsFrontend
program_arguments = 
dock_user_windows = False
pgzero_mode = False
auto_cd = True
birdseye_port = 7777
run_in_terminal_python_repl = False
run_in_terminal_keep_open = True
working_directory = C:\Users\gotoj\Documents\Grade 11 Python\Unit 1 - Python Basics\Program 1 Python Basics

[assistance]
disabled_checks = []
open_assistant_on_errors = True
open_assistant_on_warnings = False
use_pylint = True
use_mypy = True

[layout]
zoomed = False
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = VariablesView
notebook_e_visible_view = None
notebook_se_visible_view = ObjectInspector
west_pw_width = 200
east_pw_width = 403
nw_nb_height = 200
sw_nb_height = 200
s_nb_height = 361
se_nb_height = 200
ne_nb_height = 200
width = 1366
height = 705
left = 1166
top = 118

[file]
current_file = C:\Users\gotoj\Documents\Grade 11 Python\Unit 1 - Python Basics\Program 1 Python Basics\E-Garvins Sales Calculator.py
open_files = ['C:\\Users\\gotoj\\Documents\\Grade 11 Python\\Unit 1 - Python Basics\\Program 1 Python Basics\\E-Garvins Sales Calculator.py']
reopen_all_files = False
recent_files = ['C:\\Users\\gotoj\\Documents\\Grade 11 Python\\Unit 1 - Python Basics\\Program 1 Python Basics\\E-Garvins Sales Calculator.py', 'C:\\Users\\gotoj\\Documents\\Grade 11 Python\\Unit 1 - Python Basics\\Program 1 Python Basics\\Program 1 - Python Basics.py', 'C:\\Users\\gotoj\\Documents\\Grade 11 Python\\Unit 1 - Python Basics\\Case Study\\Case Study practice\\Case Study Practice Q3.py', 'C:\\Users\\gotoj\\Documents\\Grade 11 Python\\Unit 1 - Python Basics\\Python Basics - Review\\Review practice\\Python Basics Review Q1 - Adrian.py', 'C:\\Users\\gotoj\\Documents\\Grade 11 Python\\Unit 1 - Python Basics\\Python Basics - Review\\Review practice\\Python Basics Review Q2.py', 'C:\\Users\\gotoj\\Documents\\Grade 11 Python\\Unit 1 - Python Basics\\Python Basics - Review\\Review practice\\Python Basics Review Q1.py']

[edit]
tab_complete_in_editor = True
tab_complete_in_shell = True
indent_with_tabs = False

[debugger]
frames_in_separate_windows = True
automatic_stack_view = True
allow_stepping_into_libraries = False
preferred_debugger = nicer

[shell]
tty_mode = True
max_lines = 1000
squeeze_threshold = 1000
auto_inspect_values = True
```

aivarannamaa, 2020-12-06:
It turns out the problem was indeed triggered by Dracula theme. The fix will be in Thonny 3.3.1 (hopefully available tonight)

closed 2020-12-06 by aivarannamaa