thonny/thonny #1471: Using python interpreter via SSH to server

rmarkscott, 2020-12-07:
I am having trouble getting this to work. The error is:

"File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/paramiko/hostkeys.py", line 372, in from_line
    raise InvalidHostKey(line, e)
paramiko.hostkeys.InvalidHostKey: ('192.168.1.28 ecdsa-sha2-nistp256 ****************.', Error('Incorrect padding'))"

My server is 192.168.170.2, I have no idea where the 1.28 comes from.

Can you help?

aivarannamaa, 2020-12-08:
Thank you for the report!

You are using public key authentication, right? With or without passphrase?

Unfortunately I don't know what is to blame. When I googled for your error message, I found https://github.com/3drobotics/solo-cli/issues/36, which offers a work-around. Maybe it works if you remove the known host and let Paramiko (Thonny's SSH library) add it back. Hopefully your other SSH client (which added it first) is happy with the one added by Paramiko.

aivarannamaa, 2020-12-08:
> You are using public key authentication, right?

I don't know why I assumed this. The error is probably not related to authentication at all.

aivarannamaa, 2021-01-16:
Have you had chance to try the suggested work-arounds?

rmarkscott, 2021-01-16:
Yes, I never got it to work.

Mark Scott, M.S., Assistant Professor, GCIA
Cybersecurity Program Coordinator
Member - The FBI’s InfraGard Program
Freed-Hardeman University | 158 E. Main Street | Henderson, TN 38340
731.989.6779

http://www.fhu.edu/cybersecurity

aivarannamaa, 2021-01-18:
> My server is 192.168.170.2, I have no idea where the 1.28 comes from.

I suppose the library is parsing the whole known_hosts file, and you have connected to 1.28 before.

Did you try removing this line from `~/.ssh/known_hosts` ?

aivarannamaa, 2021-02-26:
Closing for now as I don't have ideas how to move forward.

closed 2021-02-26 by aivarannamaa

ollav, 2021-07-14:
> > My server is 192.168.170.2, I have no idea where the 1.28 comes from.
> 
> I suppose the library is parsing the whole known_hosts file, and you have connected to 1.28 before.
> 
> Did you try removing respective line from `~/.ssh/known_hosts` ?

I think you are right about known_hosts. I had the same issue with 'incorrect padding' and found I had a key from a very old version of ssh in ~/.ssh/known_hosts. When I removed the key the issue went away.

plijnzaad, 2023-06-28:
I can attest that removing ~/.ssh/known_hosts works (which is weird because I had already removed all keys that looked even a bit like they could be the offending ones, based on name and type)