thonny/thonny #1472: Thonny not working after Raspbian update/upgrade

rcayot, 2020-12-08:
recently having trouble with thonny in regards to Adafruit_Led_Animations library.

When loading a program onto my microcontroller board, the program will run, but a continuous "False" is returned back to the shell window.  This causes the program to be very unresponsive.

When I load my own programs the issue is not there.  If I just copy the code ( as required by circuitpython) as code.py to the microcontroller, the program runs fine.  So, it is a combination of running the programs from Thonny, and the Adafruit library.

attached is an error log when the program issued one.  It is not always generated, and it might have been generated due to the repeatedly hitting the Stop/Reset button!

I have uninstalled and reinstalled both the libraries, and Thonny a few times.

Roger

[animate-errorlog.txt](https://github.com/thonny/thonny/files/5656256/animate-errorlog.txt)

aivarannamaa, 2020-12-08:
Thank you for the report!

Could you please also post your code? Which Thonny version are you using?

rcayot, 2020-12-08:
I do n ot know which version as it does not appear to be available within the IDE.  However, it was installed with the latest update/upgrade to my Debian OS.

The code appears to be any of the examples or libraries from the adafruit_led_animations library.  I have attached a simple example which causes the problem, and another program of my own which works without the issue.  of all of my programs, only the ones from this library have the issue.

the two called modules for the animation are also attached. 
[_init_.py.txt](https://github.com/thonny/thonny/files/5659989/_init_.py.txt)

`""
This simpletest example displays the Blink animation.

For NeoPixel FeatherWing. Update pixel_pin and pixel_num to match your wiring if using
a different form of NeoPixels.
"""
import board
import neopixel
from adafruit_led_animation.animation.blink import Blink
from adafruit_led_animation.color import RED

# Update to match the pin connected to your NeoPixels
pixel_pin = board.A0
# Update to match the number of NeoPixels you have connected
pixel_num = 16

pixels = neopixel.NeoPixel(pixel_pin, pixel_num, brightness=0.5, auto_write=False)

blink = Blink(pixels, speed=0.5, color=RED)

while True:
    blink.animate()`

[color.py.txt](https://github.com/thonny/thonny/files/5659918/color.py.txt)
[blink.py.txt](https://github.com/thonny/thonny/files/5659951/blink.py.txt)
[_init_.py.txt](https://github.com/thonny/thonny/files/5659992/_init_.py.txt)
[animation_init_.py.txt](https://github.com/thonny/thonny/files/5660003/animation_init_.py.txt)

aivarannamaa, 2020-12-10:
I assume you are using Thonny 3.3.0 (you can check it from "About => Version", but the menu is only visible in regular mode, not in simple mode, which is default in RPi)

Most likely it has to do with a bug, which causes Thonny to output the value of each "bare" expression (`blink.animate()` in your case). The bug is fixed in Thonny 3.3.1, but it may take a while until RPi updates Thonny in its repos.

Until then I can offer a work-around -- saving the result of `blink.animate()` to a dummy variable, eg. `_ = blink.animate()`

aivarannamaa, 2020-12-10:
Related to #1441

aivarannamaa, 2021-01-16:
Should be fixed in Thonny 3.3.2

closed 2021-01-16 by aivarannamaa