thonny/thonny #1474: Telemetry / sending keystrokes and programs?

danbst, 2020-12-08:
For my class I'd like to make my students be able to run Thonny in "telemetry"-mode: keystrokes and editor code are collected and sent to telemetry server, which I can then render in browser together with editor (student) ID.

This particular idea came after reading paper http://ceur-ws.org/Vol-2734/paper8.pdf (cc @acbart), which mentions that "The  Thonny  environment  was  not  instrumented  to  collect log data".

And browser rendering is actually an idea from https://roc-hci.com/wp-content/uploads/Codeopticon-one-to-many-tutoring-interface_UIST-2015.pdf (сс @pgbovine), the many-to-one display, not the chat.

Would such telemetry plugin be an acceptable contribution?

acbart, 2020-12-09:
I think I wrote that sentence unclearly. The Thonny environment was not configured to collect and deliver their logs that semester - it does actually have logging, though I haven't taken advantage of it in my own research. Too tough to get the logs off the students' computers. It'd be nice to have a way to get that log data, it's a bit of a gap in my analysis.

danbst, 2020-12-09:
@acbart oh indeed, there is some "replayer" and "event_logger" functionality, though it seems to not be used actively.

aivarannamaa, 2020-12-09:
Good idea!

I recommend you start with https://github.com/thonny/thonny/wiki/User-action-logs 

The built-in plug-in which does the logging is here: https://github.com/thonny/thonny/blob/master/thonny/plugins/event_logging.py

It shouldn't be too hard to use this example for creating a [3rd party plug-in](https://github.com/thonny/thonny/wiki/Plugins), which sends the data to the network instead.

aivarannamaa, 2020-12-09:
If you're interested in creating a general solution, then I'd be happy to include it in Thonny 4.0 as a built-in plugin. 

In a general solution I imagine a menu item, which allows the user to log-in to the service by providing server url, username and password.

closed 2021-02-26 by aivarannamaa