thonny/thonny #1477: Add proxy setting

frank4466, 2020-12-11:
Hi,
Is it possible to add proxy management.
Thank you

aivarannamaa, 2020-12-11:
In Thonny 3.3 you can set `https_proxy` (or `http_proxy`) environment variables in "Tools => Options => General".

Let me know if this works for you!

aivarannamaa, 2020-12-11:
Related to #333 and #535

frank4466, 2020-12-14:
I have add in general this command and unable to download python lib.
```
set http_proxy=http://proxy.myproxy.com
set https_proxy=https://proxy.myproxy.com
```
But I'm on a PC that had a local version of Thonny and I'm now running the portable version. On the portable version the target folder is in AppData not in the thonny portable folder.
It may be related. How to change it?

aivarannamaa, 2020-12-14:
If you added these lines to Tools => Options => General, then just remove `set `-s.

frank4466, 2020-12-14:
i tried too. It does not work.
```
http_proxy=http://proxy.myproxy.com
https_proxy=https://proxy.myproxy.com
```

aivarannamaa, 2020-12-14:
Do you see your variables in `os.environ`:

```
>>> import os
>>> os.environ["http_proxy"]
???
```

If yes, then when exactly do you get the error? When searching for the package?

frank4466, 2020-12-16:
It works with the installed version but not with the portable version.

MrHarcombe, 2021-11-20:
We are using Thonny in a school environment with an autodiscover "PAC" proxy (I think it's running "Smoothwall" as a brandname, if that helps) but that makes getting hold of a specific host/port a bit of a challenge.

From experimenting around in code, I have found that the [PyPAC](https://pypi.org/project/PyPAC/) package works beautifully, but I realise that would add a bit of an overhead as you would definitely need a UI option to check if the user wants to use an autodiscover proxy and then all requests would have to be done through a proxy-aware session rather than just through the normal HTTP request mechanism you've been using up until now.

Hopefully that all makes sense - it seemed to fit as an extra comment here, but I'm happy to raise it as a separate issue (for autodiscover proxies) if you feel it's a separate topic?

aivarannamaa, 2021-12-03:
@MrHarcombe, sorry for the late reply!

I'll see whether I can use https://docs.python.org/3/library/urllib.request.html#urllib.request.getproxies for querying proxy settings from the OS level.

I'd like to avoid creating a configuration page for this.

MrHarcombe, 2021-12-03:
That works for me! Something that "just works" in school would be ideal.
I've got separate arguments to have about student access *through* the
proxy, but making Thonny transparently access the proxy would be brilliant.

---
Ian Harcombe