thonny/thonny #1480: thonny3.3.0 ESP8266 port com ERROR   thonny.plugins.micropython.backend: Crash in backend

gxlzwq, 2020-12-12:
ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 44, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "/usr/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 130, in __init__
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 44, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "/usr/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 130, in __init__
    self._prepare(clean)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 149, in _prepare
    self._prepare_helpers()
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 196, in _prepare_helpers
    self._execute_without_output(script)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 544, in _execute_without_output
    out, err = self._execute(script, capture_output=True)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 518, in _execute
    self._execute_with_consumer(script, consume_output)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 548, in _execute_with_consumer
    self._submit_code(script)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 524, in _submit_code
    self._connection.read_all_expected(expected_echo, timeout=WAIT_OR_CRASH_TIMEOUT)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 115, in read_all_expected
    actual = self.read(len(expected), timeout=timeout)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 49, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'\nclass __thonny_helper:\n    try:\n        import uos as os\n    except ImportError:\n        import os\n    import sys\n\n    # for object inspector\n    inspector_values = dict()\n    last_repl_values = []\n    @classmethod\n    def print_repl_value(cl')
    self._prepare(clean)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 149, in _prepare
    self._prepare_helpers()
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 196, in _prepare_helpers
    self._execute_without_output(script)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 544, in _execute_without_output
    out, err = self._execute(script, capture_output=True)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/backend.py", line 518, in _execute
    self._execute_with_consumer(script, consume_output)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 548, in _execute_with_consumer
    self._submit_code(script)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 524, in _submit_code
    self._connection.read_all_expected(expected_echo, timeout=WAIT_OR_CRASH_TIMEOUT)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 115, in read_all_expected
    actual = self.read(len(expected), timeout=timeout)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/connection.py", line 49, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'\nclass __thonny_helper:\n    try:\n        import uos as os\n    except ImportError:\n        import os\n    import sys\n\n    # for object inspector\n    inspector_values = dict()\n    last_repl_values = []\n    @classmethod\n    def print_repl_value(cl')

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2021-01-16:
I'm sorry for the late reply -- somehow I had forgotten about this issue.

Please try newest Thonny version.

gxlzwq, 2021-01-19:
Thonny3.3.2 support esp8266 webrepl. well done. @aivarannamaa

closed 2021-01-19 by gxlzwq