thonny/thonny #1482: Some number literals are not highlighted correctly

r0the, 2020-12-12:
The syntax highlighting of certain number literals does not work correctly:

[image]

Expected behaviour: All number literals are highlighted completely, including:
- binary integer literals
- octal integer literals
- hexadecimal integer literals
- `j` in imaginary literals
- literals with `_` for grouping purposes

All examples are from the [Python 3 documentation](https://docs.python.org/3/reference/lexical_analysis.html#integer-literals).

r0the, 2020-12-12:
If I'm guessing right, the regex to parse literals is [here](https://github.com/thonny/thonny/blob/3aa2d830b4e9af6a3510abfdc8d2f7a3b4550ab9/thonny/token_utils.py#L22).

Can I suggest the following regex to solve this issue (not very thoroughly tested):
``` python
NUMBER = matches_any("number", [
    r"\b0[bB][_0-1]+",
    r"\b0[oO][_0-7]+",
    r"\b0[xX][_0-9a-fA-f]+",
    r"\b([_\d]+(\.[_\d]*)?|\.[_\d]+)([eE][+-]?[_\d]+)?",
    ]
)
```

aivarannamaa, 2020-12-13:
Thank you for the report!

I never thought of it but you are right -- those literals should also be properly colored. I'll test your regexes and change if required. Thank you!

r0the, 2020-12-13:
I only realised it after looking at some CircuitPython code and wondering whether Python really supports binary literals ;-).

I just noticed that my suggestion didn't cover imaginary number literals. Adding `j?` should do the trick:

```
r"\b([_\d]+(\.[_\d]*)?|\.[_\d]+)([eE][+-]?[_\d]+)?j?"
```

r0the, 2020-12-13:
Btw, thank you very much for your ongoing work on Thonny. It's a great tool both for teaching and for working with microcontrollers.

closed 2021-01-05 by aivarannamaa