thonny/thonny #1485: Change tab color with interpretor

frank4466, 2020-12-14:
Hi,
is it possible to change interpreter in each tab and change color with interpreter. Or open two instances of Thonny one for microbit and one for pc, each one with different background color.

For example i will code for a micro:bit application wich communicate with PC application.

aivarannamaa, 2020-12-14:
Nope, this is not possible. 

You can open two instances of Thonny (see the first option in "Tools => Options => General"), but there is no different styling and the last instance to be closed overwrites the configuration for both.

(Most likely I'll never add the possibility to use two different interpreters in the same instance, because Thonny is an will remain a beginner's tool)

frank4466, 2020-12-16:
And allow to define one color (the background or the toolbar background for example) according to the python interpreter in order to see directly on which one is. My students are beginners too and the more visual the easier it is.

aivarannamaa, 2020-12-16:
I think Thonny 3.3 has a nice alternative -- the label of the back-end in the lower-right corner of the main window. Another clear indicator is the file browser (View => Files) -- with local CPython it has single pane, with MicroPython it has two.

Allowing the user to define colors for the back-end doesn't seem important enough for me.

closed 2021-01-16 by aivarannamaa