thonny/thonny #1487: Thonny micropython W600 602 errors.

Flaming-crossbow, 2020-12-16:
I am trying to get A W600 thumb drive with micropython running but i keep getting errors in Thonny.
Anybody got an idea what went wrong here?

MicroPython v1.10-282-g6a9b3cb-dirty on 2019-09-17; WinnerMicro module with W600
Type "help()" for more information.
ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 130, in __init__
    self._prepare(clean)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 149, in _prepare
    self._prepare_helpers()
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 196, in _prepare_helpers
    self._execute_without_output(script)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 544, in _execute_without_output
    out, err = self._execute(script, capture_output=True)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 518, in _execute
    self._execute_with_consumer(script, consume_output)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 548, in _execute_with_consumer
    self._submit_code(script)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 524, in _submit_code
    self._connection.read_all_expected(expected_echo, timeout=WAIT_OR_CRASH_TIMEOUT)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 117, in read_all_expected
    assert expected == actual, "Expected %r, got %r" % (expected, actual)
AssertionError: Expected b'\nclass __thonny_helper:\n    try:\n        import uos as os\n    except ImportError:\n        import os\n    import sys\n\n    # for object inspector\n    inspector_values = dict()\n    last_repl_values = []\n    @classmethod\n    def print_repl_value(cls, obj):\n   ', got bytearray(b'\r\nclass __thonny_helper:\r\n    try:\r\n        import uos as os\r\n    except ImportError:\r\n        import os\r\n    import sys\r\n\r\n    # for object inspector\r\n    inspector_values = dict()\r\n    last_repl_values = []\r\n    @classmethod\r\n    def print_repl_value(cls')
Traceback (most recent call last):
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 130, in __init__
    self._prepare(clean)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 149, in _prepare
    self._prepare_helpers()
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 196, in _prepare_helpers
    self._execute_without_output(script)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 544, in _execute_without_output
    out, err = self._execute(script, capture_output=True)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 518, in _execute
    self._execute_with_consumer(script, consume_output)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 548, in _execute_with_consumer
    self._submit_code(script)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 524, in _submit_code
    self._connection.read_all_expected(expected_echo, timeout=WAIT_OR_CRASH_TIMEOUT)
  File "C:\Users\a\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 117, in read_all_expected
    assert expected == actual, "Expected %r, got %r" % (expected, actual)
AssertionError: Expected b'\nclass __thonny_helper:\n    try:\n        import uos as os\n    except ImportError:\n        import os\n    import sys\n\n    # for object inspector\n    inspector_values = dict()\n    last_repl_values = []\n    @classmethod\n    def print_repl_value(cls, obj):\n   ', got bytearray(b'\r\nclass __thonny_helper:\r\n    try:\r\n        import uos as os\r\n    except ImportError:\r\n        import os\r\n    import sys\r\n\r\n    # for object inspector\r\n    inspector_values = dict()\r\n    last_repl_values = []\r\n    @classmethod\r\n    def print_repl_value(cls')

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2020-12-16:
It's quite similar to #1461. It looks like with some boards (or with some USB drivers?) the paste mode in the REPL doesn't work as it should. I'll look into this.

aivarannamaa, 2020-12-16:
Oh, this one is actually a bit different. W600 seems to add extra `\r`-s in the echo.

Flaming-crossbow, 2020-12-26:
> It's quite similar to #1461. It looks like with some boards (or with some USB drivers?) the paste mode in the REPL doesn't work as it should. I'll look into this.

Did you manage to find out or solve anything?

aivarannamaa, 2020-12-28:
> Did you manage to find out or solve anything?

Nope, not yet.

aivarannamaa, 2021-01-05:
Fixed under #1516, will be released in Thonny 3.3.2 (hopefully in a week)

closed 2021-01-05 by aivarannamaa