thonny/thonny #1488: Thonny RepeatedlyCrashing on Mac OS

michaelayu, 2020-12-16:
[thonny_crash_problem_report.txt](https://github.com/thonny/thonny/files/5705725/thonny_crash_problem_report.txt)

Hi there,

I hope you are doing a well! I am student who is using Thonny for a beginners programming course. Recently (over the last day or two), Thonny has been repeatedly unexpectedly crashing. I have tried updating the application, deleting and redownloading, but to no avail, it continues to crash. 

I am using a Mac in version 10.14.6. 

I was wondering if anyone would know any possible solutions for this issue? I have attached the problem report that appears every time the application crashes.

Thank you!

aivarannamaa, 2020-12-17:
Thank you for the report!

Unfortunately such crashes are probably bugs in the underlying framework (Tk) and therefore not directly under my control. What I can do, is publish another variant of Thonny 3.3.1 bundled with a newer Python and Tk. I'll try to do this before the end of the week.

aivarannamaa, 2020-12-17:
@michaelayu, please try https://github.com/thonny/thonny/releases/download/v3.3.1/thonny-3.3.1_alt.pkg with Python 3.9 and newer Tk

michaelayu, 2020-12-17:
Thank you so much for the quick response! I will install the new application and let you know how it works.

michaelayu, 2020-12-17:
Hi! The new version you provided was working for a few hours, but it has started crashing again. I have attached the problem report that I see now. [thonny_crash_problem_report2.txt](https://github.com/thonny/thonny/files/5712397/thonny_crash_problem_report2.txt)

Don't worry if you are not able to fix this issue! I can use another IDE in the meantime :)

Thank you for your help and efforts!

aivarannamaa, 2020-12-17:
Do you remember changing anything in your system before the crashing started?

Somebody has suspected it has to do with switching monitors (https://core.tcl-lang.org/tk/tktview?name=4121b417d9). Are you using a separate monitor or a laptop screen?

michaelayu, 2020-12-17:
No to both - I am only using my MacBook laptop screen. 

I also didn't change anything in my system before the crashing started. The only somewhat recent change I can think of is that my laptop had a system update, but that was in late November.

aivarannamaa, 2020-12-18:
In this case, can you relate the crashes to an action you did in Thonny? Were you typing or running your code?

michaelayu, 2020-12-21:
Most of time when the crashing occurred, I was just writing code.

However, over the last few days, I have been using Thonny and it hasn't been crashing! I'm not exactly sure what happened, but I managed to finish my final project for my class without any crashing issues. I guess the new version worked! 

Thank you so much for your help!

closed 2020-12-21 by michaelayu