thonny/thonny #1490: Showing Backend terminated or disconnected. Use 'Stop/Restart' to restart. plz help

ritviksreeram, 2020-12-19:


ritviksreeram, 2020-12-19:
I have reinstalled thonny but its showing the same

aivarannamaa, 2020-12-19:
I need more details in order to help you:

* What is your Thonny version?
* Which operating system?
* When does this error appear? Right after starting Thonny or after running a program?
* Did this problem appear right after you installed Thonny or you were able to use it properly for a while?

ritviksreeram, 2020-12-20:
Version: 3.3.1
Windows 7 operating system
It appears immediately after opening thonny
I used it for a while and after a month or so I got this error
I reinstalled it before I used 2. something

aivarannamaa, 2020-12-20:
Is there anything else in the error message?

Can you go to "Tools => Open Thonny data folder"? I need configuration.ini and log-files from this folder in order to diagnose it further.

ritviksreeram, 2020-12-20:
Nothing else in the error message

[general]
configuration_creation_timestamp = 2020-11-07T22:24:39.504697
language = en_US
single_instance = True
event_logging = False
disable_notification_sound = False
debug_mode = False
ui_mode = regular
scaling = default
font_scaling_mode = default

[view]
full_screen = False
maximize_view = False
shellview.visible = True
assistantview.visible = False
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = False
outlineview.visible = False
variablesview.visible = False
show_program_arguments = False
show_plotter = False
plotter_width = 808
shell_sash_position = 400
editor_font_size = 12
io_font_size = 10
name_highlighting = False
locals_highlighting = False
paren_highlighting = True
syntax_coloring = True
highlight_current_line = False
show_line_numbers = True
recommended_line_length = 0

[file]
recent_files = ['D:\\Ritvik\\aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa.py', 'D:\\Ritvik\\q.py', 'D:\\Krish\\computer\\program.py']
current_file = D:\Ritvik\aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa.py
open_files = ['D:\\Ritvik\\aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa.py']
reopen_all_files = False

[run]
working_directory = D:\Ritvik
program_arguments = 
dock_user_windows = False
pgzero_mode = False
auto_cd = True
birdseye_port = 7777
run_in_terminal_python_repl = False
run_in_terminal_keep_open = True

[layout]
zoomed = True
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = None
notebook_e_visible_view = None
notebook_se_visible_view = None
west_pw_width = 200
east_pw_width = 386
nw_nb_height = 200
sw_nb_height = 200
s_nb_height = 169
se_nb_height = 200
ne_nb_height = 200

[edit]
tab_complete_in_editor = True
tab_complete_in_shell = True

[debugger]
frames_in_separate_windows = True
automatic_stack_view = True
allow_stepping_into_libraries = False
preferred_debugger = nicer

[shell]
tty_mode = True
max_lines = 1000
squeeze_threshold = 1000

[assistance]
friendly_traceback_level = 0
open_assistant_on_errors = True
open_assistant_on_warnings = False
use_pylint = True
use_mypy = True

aivarannamaa, 2020-12-20:
configuration.ini doesn't show anything suspicious. Could you please also send the log-files from the same folder?

ritviksreeram, 2020-12-20:
backend : INFO    thonny: Thonny version: 3.3.1
backend_faults: (none)
frontend : INFO    thonny: Thonny version: 3.3.1
frontend_faults : (none)

aivarannamaa, 2020-12-20:
Strange. The only thing left is to delete configuration.ini (while Thonny is closed) and then open Thonny again. This way you get the "factory settings".

If this works, then please close Thonny again and send me configuration.ini again so that I can find the difference.

If it doesn't work then please try to remind whether you changed anything in your system before this problem started.

ritviksreeram, 2020-12-20:
[general]
configuration_creation_timestamp = 2020-12-20T12:35:16.560758
language = en_US

[view]
full_screen = False
maximize_view = False
shellview.visible = True
assistantview.visible = False
astview.visible = False
stackview.visible = False
exceptionview.visible = False
filesview.visible = False
heapview.visible = False
helpview.visible = False
notesview.visible = False
objectinspector.visible = False
outlineview.visible = False
variablesview.visible = False
show_program_arguments = False
show_plotter = False

[layout]
zoomed = True
notebook_nw_visible_view = None
notebook_w_visible_view = None
notebook_sw_visible_view = None
notebook_s_visible_view = ShellView
notebook_ne_visible_view = None
notebook_e_visible_view = None
notebook_se_visible_view = None
west_pw_width = 200
east_pw_width = 200
nw_nb_height = 200
sw_nb_height = 200
s_nb_height = 200
se_nb_height = 200
ne_nb_height = 200

[file]
current_file = None
open_files = []

[run]
program_arguments = 
dock_user_windows = False
pgzero_mode = False

ritviksreeram, 2020-12-20:
still showing the same Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2020-12-20:
Could you please send me the screenshot of your Thonny window? Just in case.

ritviksreeram, 2020-12-20:
[image]

aivarannamaa, 2020-12-20:
Have you installed any extra packages for Thonny? 

Please open Windows Explorer and paste following to the address bar: `%APPDATA%\Python\Python37\site-packages`. Which packages do you see? You can send me the screenshot of this folder

ritviksreeram, 2020-12-20:
[image]

aivarannamaa, 2020-12-20:
If you don't remember changing anything in your system before the problem started, then I'm out of ideas, sorry.

ritviksreeram, 2020-12-20:
Thank you for your help

closed 2020-12-20 by ritviksreeram

aivarannamaa, 2020-12-20:
BTW. If you want to continue using Python (eg. with another IDE), then I recommend upgrading to Windows 10. Latest Python version (3.9) does not run in Windows 7 at all anymore.

sahabitukur01, 2022-03-01:
Hello @ritviksreeram and @aivarannamaa, I encounted thesame problem,but i just went to "select interpreter" and selected the right interpreter.I think that might work

Neylajb, 2022-05-12:
Did you found a solution please? i'm struggling with the same thing

scurv, 2022-06-04:
I'll run into this when there is more than one thonny installed, 

I ended up working around it with making a bat file to start thonny. 

`@echo off

set PATH=X:\python310\.thonny\Scripts
call "X:\python310\.thonny\Scripts\activate.bat"
start pythonw.exe -m thonny`

Some packages get really temperamental with python if you attempt to use them and they are installed in more than one place on path, such as your %Users folders site packages and in python's site packages.  And python venv's often write in the hard path to your venvs address in their pyvenv.cfg. So I'll often make a vhd to run my python from if I intend to use it in more than one location. 

Otherwise you start ending up with issues like this, although this might not be the cause of it in this situation, but it was in a few of mine.