thonny/thonny #1496: Ctrl+Click on method fails: __init__() got an unexpected keyword argument 'line'

yossioo, 2020-12-28:
The title summarizes it:
I've installed Python3.9 on Ubuntu18.04 **alongside** the default 3.6.9, and installed _thonny_ in custom venv.

Having the following lines in the editor:
```
import os
print(os.path.expanduser("~"))
```

Trying to Ctrl+Click on the `expanduser` method yields the following error:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/tkinter/__init__.py", line 1884, in __call__
    return self.func(*args)
  File "/home/yossi/Dev/main_python_venv_3.9/lib/python3.9/site-packages/thonny/plugins/goto_definition.py", line 21, in goto_definition
    script = Script(source, line=line, column=column, path="")
TypeError: __init__() got an unexpected keyword argument 'line'

```
Please comment if additional information is required.

Thonny 3.3.1
Python 3.9.1 x64
Ubuntu 18.04
Tk 8.6.8

aivarannamaa, 2020-12-28:
Thank you for the report!

It looks like API of a dependency has changed. I'll fix this before the next release.

aivarannamaa, 2021-01-03:
Fixed under #1497

closed 2021-01-03 by aivarannamaa