thonny/thonny #1497: Autocomplete error

yossioo, 2020-12-28:
Pressing tab on `prin` should put `print` , but instead got this:
[image]

Thonny 3.3.1
Python 3.9.1 x64
Ubuntu 18.04
Tk 8.6.8

aivarannamaa, 2020-12-28:
Thank you for the report!

I assume you installed Thonny with `pip`, is it right? Could you please check the version of `jedi` library your Thonny using? (Tools => Manage plug-ins => Select "jedi" from the list in the left)

yossioo, 2020-12-28:
That is correct. I have a testing `venv` environment derived from system-wide python3.9 installed from sources.
In the said environment I have installed Thonny using `python3 -m pip install -U thonny` command.

The `jedi` version is 0.18.0
[image]

aivarannamaa, 2020-12-28:
Please try downgrading jedi to 0.17 (either with pip or via Thonny's plugin-manager dialog)

yossioo, 2020-12-29:
That seems to solve the problem!

aivarannamaa, 2020-12-29:
Good to know. Thanks!

closed 2021-01-03 by aivarannamaa