thonny/thonny #1504: 'simple' problem with printing 'f-string'

jluscher, 2021-01-01:
I've been debugging a problem that turned up while testing a section of a larger program with thonny.
I've spent a considerable amount of time ... partitioning ... the code to isolate the problem, but now that
I have it down to 5 lines, well I just don't know where to go !!
the code that doesn't work:

print("test: Thonny formatted printing? fail ")
for i in range(50):
_   print(f"### {i}")
print("########## test ended ##########")
exit()

Thonny gives this message after the 24th pass through the loop:
"""...
    ### 23
    ### 24
    Backend terminated or disconnected. Use 'Stop/Restart' to restart.
"""
If the "exit()" line is commented out the loop counter runs to 49, prints "#### test ended ####'
and quits happily.
When it fails it dies BEFORE it ever reaches that "test ended" print line.... !!!

I hope this makes sense to someone else as I'm discombobulated, flummoxed, confounded, etc. ;-) ...

[Thonny_printing_problem.zip](https://github.com/thonny/thonny/files/5758567/Thonny_printing_problem.zip)

aivarannamaa, 2021-01-01:
Thank you for this thorough report!

Does the same happen even after you close and re-open Thonny? Is it always `24` that gets printed last? What if you duplicate the print-statement on the first line? Or when you duplicate the print-statement under `for`?

aivarannamaa, 2021-01-01:
I was able to reproduce this. The fix will be in the next release (but I don't know when the next release will end up in Ubuntu repo).

closed 2021-01-01 by aivarannamaa

jluscher, 2021-01-02:
THANK YOU aivarannamaa !
I'm sorry for failure to respond to your questions, but glad my documentation of "problem" (NOT with 'f-string" :-( ) was useful.  
I just turned 74 years 29th, and my grandson turned 1 year on 31st (ZOOM), and New Years ;-)  
... so I was slow to realize that there could be a 'TIMING' problem with an output queue. ;-) 
... and in the meantime you figured it out and fixed it. !!  
Nicely done.  
Grateful Applause !!

aivarannamaa, 2021-01-02:
Until I release Thonny 3.3.2, you could try `sys.exit` instead of built-in `exit`. For some reason `sys.exit` doesn't seem to trigger this problem (at least in my tests), and it is recommended over built-in `exit` in scripts anyway (https://stackoverflow.com/a/6501134/261181).

jluscher, 2021-01-03:
Thanks, will do !!