thonny/thonny #1505: How to add an interpreter to thonny

Quinniboi10, 2021-01-01:
Hello, I really like Thonny, but I want to add an interpreter so I can run python 2.7 files. Any help is welcome. 

Thanks,
Quinniboi10

aivarannamaa, 2021-01-01:
You can add a Python 3 interpreter from "Run => Select interpreter => Custom Python ...", but Python 2 is not supported.

Out ouf curiosity -- why do you need Python 2?

Quinniboi10, 2021-01-02:
To use the GMailAPI, if I use C:\Users\<MYUSERNAME>\anaconda3\ipython.exe, will I get to use all of the libraries that I installed when using Spyder (from anaconda3)?

aivarannamaa, 2021-01-02:
> if I use C:\Users<MYUSERNAME>\anaconda3\ipython.exe, will I get to use all of the libraries that I installed when using Spyder (from anaconda3)?

That's right.

aivarannamaa, 2021-01-02:
Oops, sorry, don't use `ipython.exe`. There should be `python.exe` in the same directory -- use this one instead

closed 2021-01-02 by aivarannamaa