thonny/thonny #1508: Connection error using thonny with cirtuitPython on esp32s2

bluejazzCHN, 2021-01-02:
Error is below:

Adafruit CircuitPython 6.0.1 on 2020-12-28; Saola 1 w/Wrover with ESP32S2

Connection lost (device reports readiness to read but returned no data (device disconnected or multiple access on port?))

and USB drive slash on Linux System

Use Stop/Restart to reconnect.
Right when using PuTTy SSH connect

aivarannamaa, 2021-01-03:
Are you sure, you didn't have the port open in another application?

> and USB drive slash on Linux System

Was this also part of the error message?

bluejazzCHN, 2021-01-04:
yes, i am sure. connect and lost. if use other terminal, Putty ssh no problem.

aivarannamaa, 2021-01-04:
I guess I'll order one for myself. AliExpress lists 4 models:

* ESP32-S2-Saola-1R
* ESP32-S2-Saola-1RI
* ESP32-S2-Saola-1M
* ESP32-S2-Saola-1MI

Which one do you have?

aivarannamaa, 2021-01-04:
Do you have psram and antenna connector?

r0the, 2021-01-04:
@bluejazzCHN Maybe you want to try [CircuitPython 6.1.0 Beta 3](https://github.com/adafruit/circuitpython/releases/tag/6.1.0-beta.3). This version is suggested for ESP32-S2 devices on the [CircuitPython](https://circuitpython.org/board/targett_module_clip_wrover/) website and fixes a number of ESP32-S2 related issues.

ESP32-S2 support has been added very recently with CircuitPython 6.0.0

bluejazzCHN, 2021-01-07:
> I guess I'll order one for myself. AliExpress lists 4 models:
> 
> * ESP32-S2-Saola-1R
> * ESP32-S2-Saola-1RI
> * ESP32-S2-Saola-1M
> * ESP32-S2-Saola-1MI
> 
> Which one do you have?

ESP32-S2-Saola-1M

aivarannamaa, 2021-01-28:
@bluejazzCHN , I now received my Saola, but I'm not able to get the REPL even outside of Thonny.

Initially I received a list of colors from it in Thonny, but could not enter REPL. Unfortunately I didn't try other tools with the initial firmware.

After I installed CircuitPython 6.1 on it (https://gist.github.com/askpatrickw/0a276c7e2d4f54e442b2cb6eaa0d32ea), I didn't get any output anymore and couldn't get to REPL neither with Thonny nor with `tio`.

Were you using the factory firmware or did you install something else? As far as I understand, MicroPython does not support ESP32-S2 yet (https://github.com/micropython/micropython/issues/5958).

bluejazzCHN, 2021-01-31:
> @bluejazzCHN , I now received my Saola, but I'm not able to get the REPL even outside of Thonny.
> 
> Initially I received a list of colors from it in Thonny, but could not enter REPL. Unfortunately I didn't try other tools with the initial firmware.
> 
> After I installed CircuitPython 6.1 on it (https://gist.github.com/askpatrickw/0a276c7e2d4f54e442b2cb6eaa0d32ea), I didn't get any output anymore and couldn't get to REPL neither with Thonny nor with `tio`.
> 
> Were you using the factory firmware or did you install something else? As far as I understand, MicroPython does not support ESP32-S2 yet ([micropython/micropython#5958](https://github.com/micropython/micropython/issues/5958)).

@aivarannamaa  thank you.  this questions is clearly. I have used others IDE to write code.

closed 2021-01-31 by bluejazzCHN