thonny/thonny #1509: Hello everyone, I am having a problem trying to connect with my m5stickC, someone knows how to start it, thanks in advance and for all the help you can give me ...

fraramon, 2021-01-02:


fraramon, 2021-01-02:
I've already tried a thousand ways and there's no way to make it work. If anyone knows how to help me, I would appreciate it. Thanks in advance.

aivarannamaa, 2021-01-02:
Do you get error messages? If yes, then please copy them here, together with stacktraces.

aivarannamaa, 2021-01-02:
Oh, sorry, I didn't notice your stacktraces before. Your problems looks similar to #1461. I have some remedy planned for next Thonny version. In the meanwhile you could try an older Thonny version: https://github.com/thonny/thonny/releases/tag/v3.2.7

fraramon, 2021-01-02:
I think the problem is one of the echoes, they do not match.

fraramon, 2021-01-02:
I have eliminated all the comparisons and exceptions and it worked, but of course this is not a solution, and also the recording still does not work.

fraramon, 2021-01-02:
I'm going to try the version you told me, and I check if it works.

fraramon, 2021-01-02:
Version 3.2.7 works. I've been testing and it works.
When will you release 3.3.2?

fraramon, 2021-01-02:
Another thing, when it is executed again, it is not waiting for the stop and generates the following error:
WARNING:root:Unexpected echo. Expected b'%Run -c $EDITOR_CONTENT\r\n', got bytearray(b'MediaTrans/TimerCam      ')

fraramon, 2021-01-02:
Example the program:

from m5stack import *
from m5ui import *
from uiflow import *
while True:
    wait_ms(1000)

aivarannamaa, 2021-01-03:
Thank you for the feedback! I hope to release 3.3.2 in two weeks.

jackho9, 2021-01-03:
Just FYI I have the same issue with M5StickC and M5StickC-Plus.  Reverting to 3.2.7

fraramon, 2021-01-03:
Can you confirm that you have this configuration?
Thonny 3.2.7 (MicroPython (ESP32))
M5StickC Plus Firware v1.7.0-plus (Modo de aplicación con temp.py) (In USB mode it doesn't work)
Windows 10.

jackho9, 2021-01-03:
Hello fraramon, yes that's exactly my configuration, with USB Serial Port (Com 3) on Windows 10 Home.

It works in USB mode for me.  As always when Thonny is started the M5StickC is busy and I need to interrupt it with Ctrl-F2.  At first it wouldn't run my usual python file because lcd had disappeared from m5stack.  But after a few tries lcd reappeared (weird, huh?) and it has worked fine since.  Before:

[image]

After:

[image]

fraramon, 2021-01-03:
You can do the following test:
1.- Burn M5stack with Firware v1.7.0-plus
2.- Select the temp.py program in M5stick-C
Try to see what happens.
Sorry about my English, I'm Spanish ... upps.

fraramon, 2021-01-03:
Reset the device with the left button.
Then quickly central button.
After 2 right click to appList.
Then right-click and select temp.py.
And when you start thonny you shouldn't have to hit Ctrl-c.
Regards.

jackho9, 2021-01-03:
So I turn it on, quickly press M5, then the side button twice to get to file list and select temp.py

It says "This file from web exec" - the USB connection to Thonny works and I am able to run my python file from there.

jackho9, 2021-01-03:
Only difference from your instructions is that I had Thonny already running when I did what you suggested

fraramon, 2021-01-03:
ok, But it is just an example with no while holding the program.

fraramon, 2021-01-03:
The important thing is to select a program that does not end in an infinite loop.

jackho9, 2021-01-03:
This is what's shown in Thonny after doing that

[image]

fraramon, 2021-01-03:
But that is already running your code.

fraramon, 2021-01-03:
if your program ends in an infinite loop, to cut it you must use Ctrl-C.

fraramon, 2021-01-03:
Ctrl-C should be run every time a program is started, but automatically, so that we can develop quickly.

aivarannamaa, 2021-01-05:
Should be fixed under #1516, will be released in Thonny 3.3.2 (hopefully in a week). Please reopen, if you are still having same problem in Thonny 3.3.2

closed 2021-01-05 by aivarannamaa

fraramon, 2021-01-05:
Ok, thank you very much. :+1:

fraramon, 2021-01-13:
3.3.2 works for me now, thanks.

macraj, 2021-02-12:
In 3.3.3 the same problem (with echoes) still exists. After CTRL+C I can upoad file

aivarannamaa, 2021-02-12:
@macraj, can you paste your error?