thonny/thonny #1510: How to install keras in Thonny

Quinniboi10, 2021-01-03:
Hello, I really like using Thonny, and I want to install keras on it but it won't get me. Any help is welcome

Thanks,
Quinniboi10

aivarannamaa, 2021-01-03:
>  but it won't get me

What do you mean exactly? Do you get an error message?

Quinniboi10, 2021-01-03:
1. I met let not get
2.  It says:
  ERROR: Failed building wheel for h5py
Failed to build h5py
ERROR: Could not build wheels for h5py which use PEP 517 and cannot be installed directly
Process returned with code 1

aivarannamaa, 2021-01-04:
You may need to upgrade pip: https://stackoverflow.com/questions/61365790/error-could-not-build-wheels-for-scipy-which-use-pep-517-and-cannot-be-installe

pip is another package you should see in "Tools => Manage packages". Or you could use command line ("Tools => Open system shell")

Quinniboi10, 2021-01-04:
Pip is at the latest version though.

aivarannamaa, 2021-01-16:
In this case I'm afraid I can't help you.

closed 2021-01-16 by aivarannamaa