thonny/thonny #1511: Argument autocompletion removes previous character

yossioo, 2021-01-03:
When pressing Tab or Ctrl-Space to autocomplete the argument name, a character prior to the argument is removed.
Expected behavior: completing the argument name without removing any characters.

https://user-images.githubusercontent.com/10714585/103475726-8d026e00-4db8-11eb-8012-ad51deba95ac.mp4

In this case, space is removed, but the problem occurs when there is no space character:

https://user-images.githubusercontent.com/10714585/103475737-a1466b00-4db8-11eb-9843-f8fb232dfe89.mp4

In this case, the syntax is impacted and the code won't run.

---- Configuration tested:
Thonny 3.3.2-dev
Python 3.9.1 x64
Ubuntu 18.04
Tk 8.6.8
Jedi 0.18.0

aivarannamaa, 2021-01-03:
Thank you for the thorough report!

I think I got it fixed. Please test and close the issue if you are happy!

yossioo, 2021-01-03:
Yup, it works just fine, thank you!

One little thing though, suggestions for auto-completion are printed in the terminal where I run thonny.
I understand that it is the result of this [line](https://github.com/thonny/thonny/commit/c531c3ca08f7bcfbfceca11d9422bf0d4c4d6f0f#diff-09544a1ee8e80a0c51d4b3e51657e83eb67354677d11344e84e07f721c8e0377R111).
I assume that the release version should not have this?
[image]

aivarannamaa, 2021-01-04:
> suggestions for auto-completion are printed in the terminal where I run thonny.

Oops. Fixed now. Thank you for reminding this!

closed 2021-01-04 by aivarannamaa