thonny/thonny #1513: Error renaming a file on the device in M5Stack (3.2.7).

fraramon, 2021-01-03:
The error is as follows:
```
Traceback (most recent call last):
  File "C:\Users\fran2\Documents\M5Stack\M5StickC\thonny-3.2.7-windows-portable\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\fran2\Documents\M5Stack\M5StickC\thonny-3.2.7-windows-portable\lib\site-packages\thonny\workbench.py", line 969, in dispatch_from_menu
    dispatch(None)
  File "C:\Users\fran2\Documents\M5Stack\M5StickC\thonny-3.2.7-windows-portable\lib\site-packages\thonny\workbench.py", line 911, in dispatch
    handler()
  File "C:\Users\fran2\Documents\M5Stack\M5StickC\thonny-3.2.7-windows-portable\lib\site-packages\thonny\code.py", line 767, in _cmd_rename_file
    os.remove(old_filename)
OSError: [WinError 123] El nombre de archivo, el nombre de directorio o la sintaxis de la etiqueta del volumen no son correctos: 'MicroPython device :: /flash/apps/aa.py'
```

aivarannamaa, 2021-01-04:
This is fixed in 3.3. I understand why you're using 3.2.7, but I'm not going to make releases in 3.2 line anymore.

closed 2021-01-04 by aivarannamaa

fraramon, 2021-01-04:
Ok, thanks and sorry.