thonny/thonny #1519: Multi-line repr and Variables panel

raenye, 2021-01-07:
When the string representation (`repr`) of an object consists of more than a single line, it is displayed incorrectly in the Variables panel; it gets occluded by the next lines.
Best solution for this IMHO could be to have a "folded" view (small `+`, click to open the fold and see more than just the first line of the representation).

Screenshot:
[image]

Thanks (and sorry for bringing up all these visual bugs, I know they're low priority)

aivarannamaa, 2021-01-07:
Thank you for the report!

Unfortunately the table widget is not very flexible in Tk -- I don't know a good idea how to add this + so that it scrolls properly. 

What you can do now, is open the Object inspector (from View menu). Now, when you click on a row in variables table, the object gets opened in the object inspector. Try a list, str, int for examples of different representations.

One improvement I can think of, is having Object Inspector open automatically when the user double-clicks in the variables view.

raenye, 2021-01-08:
I think it's OK not to display everything; perhaps trim each `repr` after the newline? to make sure it doesn't "invade" next rows?

aivarannamaa, 2021-01-08:
> perhaps trim each repr after the newline? to make sure it doesn't "invade" next rows?

Good idea!