thonny/thonny #1520: Allow REPL during debugging

stevenbell, 2021-01-08:
Is there a way to access the REPL while debugging a script?  A common workflow for me is to execute some code in a script up to a breakpoint, then poke around in the REPL until I figure out what's wrong or what code actually needs to be written.

I tried just putting in a call to `IPython.embed()`, but caused Thonny to stop running my code with an `OSError: [Errno 25] Inappropriate ioctl for device`.

stevenbell, 2021-01-12:
I found that `IPython.embed()` works if you run it in a terminal (Ctrl-T) instead of the regular "run".  This doesn't integrate with the GUI and breakpoints (and your code will crash if you run it inside of Thonny), but it's a usable workaround.

aivarannamaa, 2021-01-12:
I agree it would be a nice feature for some users, but it is not clear how avoid confusing beginners with it.

If you are happy to insert something into your code, then have you tried using built-in [breakpoint()](https://www.python.org/dev/peps/pep-0553/) in your code and running it with F5? When using Shift+F5 or Ctr+F5, Thonny interprets it as just a breakpoint, but it may be worth to change this, such that it breaks into pdb in both cases (running and debugging).

For exploring data, there is Object Inspector (you can double-click on an attribute to dig deeper).

I agree these features are not easy to discover. I'll add another issue for this.

aivarannamaa, 2021-01-12:
... another option is to offer something simpler instead of pdb with `breakpoint()`, eg something similar to regular Python REPL, with `>>> ` prompt for better familiarity. We don't need most of pdb commands in Thonny's context anyway.

aivarannamaa, 2021-01-12:
... alternatively, I could introduce another command (eg. toolbar button and/or shortcut), which would invoke the debug REPL regardless of how the program was stopped.

stevenbell, 2021-01-13:
Ah, thanks, `breakpoint()` integrates much more nicely than `IPython.embed()`.

I think my ideal solution would be to have a visually different prompt appear any time execution is stopped.  This would make it clear that it's not a "normal" REPL prompt, but be immediately accessible without the need to click another menu button or shortcut.  Suppose we have a simple script like this, with a breakpoint set on the second line:
``` python
x = 5
print(f"The value of x is {x}.")
```

So a debugging session might look like the following:

[user clicks "debug" button and script starts running]
```
>>> %Debug myscript.py
```
[code hits a breakpoint, and user is presented with a REPL prompt]
```
(DEBUG) >>> print(x)
5
(DEBUG) >>> x*x
25
(DEBUG) >>> x = 0
```
[user clicks continue, and script finishes running]
```
The value of x is 0.
>>>
```
[User is returned to the regular REPL.]

It would be super cool if using pdb commands at the debug REPL could feed back to the GUI debugger (e.g., typing 'step' would actually step things forward in the GUI), but I'm guessing that isn't feasible without major changes to the code.

aivarannamaa, 2021-01-15:
Printing a prompt on each stop would be useful if you don't stop too often, but beginners are more likely to wander around and make many steps to learn the code behavior. This would litter the output with debugger prompts.

On the other hand, if the debug REPL was optional and if Thonny removed the unused prompt after resuming the program, then I might like it. I need to give it more thought.

> It would be super cool if using pdb commands at the debug REPL could feed back to the GUI debugger (e.g., typing 'step' would actually step things forward in the GUI), but I'm guessing that isn't feasible without major changes to the code.

It probably won't bee too hard, but I don't see why someone would prefer typing `step` (or even something shorter) instead of pressing F6 or F7.

aivarannamaa, 2021-01-16:
A related discussion with another idea about handling `breakpoint()`: #1388

stevenbell, 2021-01-28:
> On the other hand, if the debug REPL was optional and if Thonny removed the unused prompt after resuming the program, then I might like it. I need to give it more thought.

I agree -- only keep the prompt if the user typed something; otherwise just hide it again as the code starts running.

> It probably wouldn't be too hard, but I don't see why someone would prefer typing step (or even something shorter) instead of pressing F6 or F7.

My thought with linking `pdb` commands and the GUI is mostly for consistency.  It would be bad if a user could step within the debugger and not have this reflected in the GUI.  But just preventing the user from controlling the debugger from within the debug prompt would also work.

amr66, 2026-08-24:
I simply want to confirm (in 2026) that the idea of using the REPL after stopping at a breakpoint feels "naturally" to me.  I use it a lot in other IDE for experimenting with statements, variables or just to see, what I can do with an active object. And I always tell my students to do so, as well. Many discussions about a program's logics or errors end in trying something in the REPL.