thonny/thonny #1521: Unable to install numba package in Thonny

soumenbasak, 2021-01-08:
I would like to install numba (https://pypi.org/project/nu... package (version 0.52.0) in Thonny. I receive the following error messages when I try to install this package.

/home/soumen/.thonny/Python36/include/site/python3.6/llvmlite
"failed with error code 1 in /tmp/pip-build-s2xk_fk9/llvmlite"

The operating system in my desktop is Ubuntu 18.04.5

Anybody can help me on this matter ?

aivarannamaa, 2021-01-08:
What's your Thonny version? I couldn't reproduce this in Thonny 3.3.1 and Ubuntu 20.04.

soumenbasak, 2021-01-08:
Thonny version is 2.1.16 and Python version 3.6.9

aivarannamaa, 2021-01-08:
I suspect this issue is not Thonny specific. Did you install Thonny with `apt`? Try running `pip3 install numba` on your system command line. If it still gives you errors, then you should take it to Numba's forum.

soumenbasak, 2021-01-08:
Yes I installed Thonny with apt. I also tried pip3 install numba. It works fine (see below)

soumen@soumen-Precision-Tower-7810:~$ pip3 install numba
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: numba in ./.local/lib/python3.6/site-packages (0.51.2)
Requirement already satisfied: llvmlite<0.35,>=0.34.0.dev0 in ./.local/lib/python3.6/site-packages (from numba) (0.34.0)
Requirement already satisfied: setuptools in ./.local/lib/python3.6/site-packages (from numba) (49.2.1)
Requirement already satisfied: numpy>=1.15 in ./.local/lib/python3.6/site-packages (from numba) (1.19.2)
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
soumen@soumen-Precision-Tower-7810:~$ sudo apt-get install thonny
[sudo] password for soumen: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
thonny is already the newest version (2.1.16-3).
The following packages were automatically installed and are no longer required:
  dkms gnu-smalltalk-common libasound2:i386 libavahi-glib1:i386 libcuda1-340
  libjpeg62:i386 libllvm8:i386 libllvm9 libllvm9:i386 libmbim-glib4:i386
  libmm-glib0:i386 libnvidia-common-390 libpolkit-gobject-1-0:i386
  libqmi-glib5:i386 libxtst6:i386
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 37 not upgraded.
soumen@soumen-Precision-Tower-7810:~$

aivarannamaa, 2021-01-08:
In this case you should be able to use it in Thonny, because apt-installed Thonny is supposed to use same Python that pip3

soumenbasak, 2021-01-08:
No, It is still not working. Any advise from you on this matter is most welcome.

aivarannamaa, 2021-01-08:
Please check "Tools => Options => Interpreter" (the last section may be also "Back-end", don't remember Thonny 2.1 anymore). If you have "Special virtual environment" selected, then try selecting your system Python 3 instead (Custom Python 3 ...).

In general, I'd recommend upgrading from Thonny 2.1. Your distro may not have newer versions, but you can apt remove it and install binary Thonny+Python bundle instead:

```bash
bash <(wget -O - https://thonny.org/installer-for-linux)
```

aivarannamaa, 2021-01-16:
I'm afraid I can't help you in this case

closed 2021-01-16 by aivarannamaa