thonny/thonny #1522: Thonny App can't import numpy with Anaconda / Windows

ethan-tau, 2021-01-08:
Problem: when launching the ThonnyApp from Windows Start menu, numpy (and similar packages) cannot be imported, although they can be imported just fine in my Anaconda Python interpreter environments.

```python
import numpy as np
```

results in 
```
ImportError:   IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!  Importing the numpy C-extensions failed.
```

Workaround: 
Launch an Anaconda-aware command prompt first (like Anaconda PowerShell Prompt), then execute ThonnyApp.exe from there, which is 
`C:\ProgramData\Anaconda3\Scripts\thonny.exe -m thonny` 
on my install.

Permanent personal fix (needs work): 
Modify the ThonnyApp shortcut so that it first calls the hooks to import the Anaconda environment variables (like library paths) before executing `thonny.exe`. 
Possible ideas: 
* copy arguments from Anaconda PowerShell Prompt shortcut (naive attempt didn't work).
* look at how QT Console gets launched with Anaconda

Permanent distribution fix (needs suggestions/work): 
* Modify the thonnyapp installer to detect Anaconda installation, similar to 
* Explore other install methods, like `conda install thonnyapp` instead of `pip install thonnyapp`

Installation: 
Windows 10
Python 3.7.19 from Anaconda
Numpy 1.19.2 (from Anaconda distribution)
Thonny 3.3.2/ThonnyApp 0.9.1 installed with `pip install thonnyapp` in Anaconda PowerShell Prompt

aivarannamaa, 2021-01-08:
Thank you for the report!

Have you tried installing Thonny + Python bundel (ie. regular install), and making it use Anaconda's Python as back-end (Run => Select interpreter) ?

ethan-tau, 2021-01-08:
I haven't tried the bundle install, but the Python interpreter is already set as the Anaconda interpreter, as the REPL window displays `Python 3.7.9 (C:\ProgramData\Anaconda3\python.exe)`. The import fails if Thonny isn't executed from within the Anaconda environment.

In the Thonny REPL, `sys.path` returns the same output for both failed and successful `import numpy` scenarios:
```python 
>>> import sys
>>> sys.path
['',
'C:\\tau_dev',
'C:\\ProgramData\\Anaconda3\\python37.zip',
'C:\\ProgramData\\Anaconda3\\DLLs',
'C:\\ProgramData\\Anaconda3\\lib',
'C:\\ProgramData\\Anaconda3',
'C:\\ProgramData\\Anaconda3\\lib\\site-packages',
'C:\\ProgramData\\Anaconda3\\lib\\site-packages\\win32',
'C:\\ProgramData\\Anaconda3\\lib\\site-packages\\win32\\lib',
'C:\\ProgramData\\Anaconda3\\lib\\site-packages\\Pythonwin']
```

aivarannamaa, 2021-01-10:
I could reproduce this with normal Thonny installation + Anaconda back-end. It looks like some directories containing libraries must be added to system's PATH.

The fix will be in the next release.

closed 2021-01-10 by aivarannamaa