thonny/thonny #1525: I have connect esp32 with usb cable to my raspberry Pi  it's shows device not connected

CoderLogy, 2021-01-10:
I connected device using thonny and thonny has also detected it but when I click open file from micropython device it shows me error
Pls help the esp32 is connected and have .bin installed on it

aivarannamaa, 2021-01-10:
Thank you for the report!

Please copy this error here. Also please tell me your ESP32 board name (model), MicroPython version and Thonny version.

CoderLogy, 2021-01-10:
I will add u. Error some my board is esp3p2 generic module wroom 32 micropython version is esp32-idf3-20200902-v1.13.bin and thonny is latest

aivarannamaa, 2021-01-16:
Unfortunately I can't reproduce this. Can you access the REPL with another tool (eg. screen, tio, minicom)?

CoderLogy, 2021-01-16:
Yes now my problem fixed thnks for reaching me

closed 2021-01-16 by aivarannamaa