thonny/thonny #1526: Badly handled errors with unknown filetypes

Ivoz, 2021-01-10:
I believe this can happen if you try to open a file from Thonny's explorer window, when it has no program association. You can get a number of problems from this:

Internal Error
```
Traceback (most recent call last):
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 562, in on_double_click
    self.open_path_with_system_app(path)
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 900, in open_path_with_system_app
    open_with_default_app(path)
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 1327, in open_with_default_app
    os.startfile(path)
OSError: [WinError 1155] No application is associated with the specified file for this operation: 'C:\\Users\\Matthew\\code\\orbits\\orbits.ico'
```

and afterwards, whenever you alt-tab back to Thonny:

Problem when handling 'WindowFocusIn'
```
Traceback (most recent call last):
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1706, in event_generate
    handler(event)
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 903, in on_window_focus_in
    self.refresh_tree()
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 742, in refresh_tree
    self.render_children_from_cache("")
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 407, in render_children_from_cache
    self.request_dirs_child_data(node_id, self.get_open_paths() | {path})
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 879, in request_dirs_child_data
    self.render_children_from_cache(node_id)
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 469, in render_children_from_cache
    self.render_children_from_cache(child_id)
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 407, in render_children_from_cache
    self.request_dirs_child_data(node_id, self.get_open_paths() | {path})
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 879, in request_dirs_child_data
    self.render_children_from_cache(node_id)
  File "C:\Users\Matthew\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 427, in render_children_from_cache
    raise RuntimeError("None data for %s" % path)
RuntimeError: None data for C:\Users\Matthew\code\orbits\orbits.ico
```

aivarannamaa, 2021-01-14:
Thank you for the report!

Unfortunately I couldn't reproduce these errors. 

I could reproduce the first one only by adding a mistake in code. For this I added a guard, so that when opening in system app causes problems, a simpler dialog gets presented, but I  don't know how to reproduce the second one.

aivarannamaa, 2021-01-14:
Later I got the second error as well, but I still don't know how to reproduce it reliably.

Ivoz, 2021-01-14:
Have you tried creating a file which has no default application to open it with (maybe doesn't know how to render a thumbnail for it), and played around in thonny?

aivarannamaa, 2021-01-16:
I tried, but Windows offered me "Open with" dialog for this, instead of WinError.

Anyway, I suspect there is another mistake causing the second error, and the first error is just enabling it. As I'm not able to reproduce it relieably, I'm closing this for now. Hopefully, when it appears again in another context, I'll get to reproduce it more easily.

closed 2021-01-16 by aivarannamaa