thonny/thonny #1529: System Shell environment is truncated on macos

bortels, 2021-01-10:
Thonny 3.3.2 on macos Big Sur - when launching "Tools->open system shell..." - the environment sent is overly long and hits line length limits, in my case truncating the path.  

`Last login: Sun Jan 10 14:25:36 on ttys001
clear; cd /Users/tbortels; export TCL_LIBRARY=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/tcl8.6; export TK_LIBRARY=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/tk8.6; export SSL_CERT_DIR=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/certifi; export SSL_CERT_FILE=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/certifi/cacert.pem; export PATH=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/bin:/Users/tbortels/Library/Python/3.7/bin:/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.7/bin:/usr/bin:/bin:/usr/sbin:/sbin; /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Pytho
➜  ~ clear; cd /Users/tbortels; export TCL_LIBRARY=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/tcl8.6; export TK_LIBRARY=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/tk8.6; export SSL_CERT_DIR=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/certifi; export SSL_CERT_FILE=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/certifi/cacert.pem; export PATH=/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/bin:/Users/tbortels/Library/Python/3.7/bin:/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.7/bin:/usr/bin:/bin:/usr/sbin:/sbin; /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/../../../../../../../Python.framework/Versions/3.7/bin/python3.7 /Applica
`

I'll add the entries in the path themselves are extra long (the ../../.. chain backtracking).

aivarannamaa, 2021-01-14:
Thank you for the report!

I'll try normalizing the PATH entries to save some space, but I don't have any other ideas.

I'm surprised to see same variables assigned twice. I assume this is a side effect of the too long line, because I can't see how Thonny could make this mistake

aivarannamaa, 2021-01-14:
Also, I could extract common prefix of the values (`/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7`) into a helper variable

aivarannamaa, 2021-01-16:
Should be now shorter in the upcoming 3.3.3 release.

closed 2021-01-16 by aivarannamaa