thonny/thonny #1530: Transparent App icon on macos is difficult to hit

bortels, 2021-01-10:
Macs are a bit brain damaged, in that the transparent areas on the icon for an app are *not* clickable. 

That means to launch the Thonny app, you have to hit on the thin bars of the "Th" letters that the app icon has - the empty space between or inside the characters isn't a valid target. 

I'd suggest an opaque background to make it easier to launch.

r0the, 2021-01-11:
@bortels Good idea, an icon with background would also better match other application icons in the dock. Currently it sticks out a bit:

[image]

@aivarannamaa Here's a proposal based on the current app icon [here](https://github.com/thonny/thonny/blob/master/thonny/res/thonny.png) and the  [Apple Design Resources](https://developer.apple.com/design/resources/#macos-apps).

[image]

Preview in dock:

[image]

But it's very white. Maybe either background should be a different colour or the letters should be filled or be
in a different colour...

aivarannamaa, 2021-01-11:
Thank you both for report and suggestion. I never noticed this problem myself.

I think the white one is perfectly good. Thanks!

r0the, 2021-01-12:
Actually, the non-clickable transparency is only a problem in the finder when the files are displayed as icons. In the dock and launcher, the transparent areas are clickable.

aivarannamaa, 2022-03-16:
I shifted the letters a bit, so that it looks a bit more balanced:
[image]

aivarannamaa, 2022-03-21:
> But it's very white. Maybe either background should be a different colour or the letters should be filled or be in a different colour...

@r0the, what do you think of making the background partly transparent?

varunkumarmedam, 2024-03-24:
Any work around for making transparent background clickable?