thonny/thonny #1539: donwload is very slow

speedy-10, 2021-01-13:
Hi, tried the download proposed on thonny.org:

bash <(wget -O - https://thonny.org/installer-for-linux)

It is very slow the last 2 evenings, it would take about 30-40 minutes.
I was just wondering.

Regards.

aivarannamaa, 2021-01-14:
Thank you for the report! Unfortunately I can't reproduce this -- for me it's about 20 seconds.

Have you checked that it's download part which is slow? Or installation part?

Is it also slow when you try to download from here: https://github.com/thonny/thonny/releases/tag/v3.3.2 ?

speedy-10, 2021-01-14:
Hi, ijust now try again:
24,2KB/s

speedy-10, 2021-01-14:
~$ bash <(wget -O - https://thonny.org/installer-for-linux)

Ausgabe wird nach »wget-log.11« umgeleitet.

This script will download and install Thonny (thonny-3.3.2) for Linux (32 or 64-bit PC).
Press ENTER to continue or Ctrl+C to cancel.
Downloading https://github.com/thonny/thonny/releases/download/v3.3.2/thonny-3.3.2-x86_64.tar.gz
--2021-01-14 13:21:34--  https://github.com/thonny/thonny/releases/download/v3.3.2/thonny-3.3.2-x86_64.tar.gz
Auflösen des Hostnamens github.com (github.com) … 140.82.121.4
Verbindungsaufbau zu github.com (github.com)|140.82.121.4|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://github-production-release-asset-2e65be.s3.amazonaws.com/163728962/873ba100-4fe1-11eb-9cad-e3d8729f0ce1?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210114%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210114T122135Z&X-Amz-Expires=300&X-Amz-Signature=4ae4972f03017fa4fdcb7fdf578bf1e28e2c514724c98d5770b183d719e301c4&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=163728962&response-content-disposition=attachment%3B%20filename%3Dthonny-3.3.2-x86_64.tar.gz&response-content-type=application%2Foctet-stream [folgend]
--2021-01-14 13:21:35--  https://github-production-release-asset-2e65be.s3.amazonaws.com/163728962/873ba100-4fe1-11eb-9cad-e3d8729f0ce1?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210114%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210114T122135Z&X-Amz-Expires=300&X-Amz-Signature=4ae4972f03017fa4fdcb7fdf578bf1e28e2c514724c98d5770b183d719e301c4&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=163728962&response-content-disposition=attachment%3B%20filename%3Dthonny-3.3.2-x86_64.tar.gz&response-content-type=application%2Foctet-stream
Auflösen des Hostnamens github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com) … 52.217.105.140
Verbindungsaufbau zu github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.217.105.140|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 32638311 (31M) [application/octet-stream]
Wird in »./tmp.S6Uv6k2Vra/thonny-3.3.2-x86_64.tar.gz« gespeichert.

6Uv6k2Vra/thonny-3.3.2  99%[==========================> ]  30,90M  17,3KB/s    ETA 9s

aivarannamaa, 2021-01-14:
The only explanation I can offer, that at your location the GitHub's download servers resolve to different machines and they are slow for some reason.

aivarannamaa, 2021-01-14:
I found this: https://github.community/t/slow-downloads-telekom-germany/154790

closed 2021-01-16 by aivarannamaa