thonny/thonny #1541: Add support for installing regular PyPI packages to MicroPython or CircuitPython

aivarannamaa, 2021-01-14:
micropip is having problems with Some PyPI libraries, because they have versioned dependencies (eg. adafruit-circuitpython-neopixel, see https://groups.google.com/g/thonny/c/YKCU4pcrcU8/m/u5DtddwkCgAJ).

Possible solutions:

* enhance micropip
* install those packages with pip (see https://stackoverflow.com/a/2916320/261181)
* Can this be done with Circup (#1540)?

dnguyen76, 2021-01-25:
Hello 
I can't import MicroWebSrv2 with tools-->modules i had imported MicroWebSrv2 with pip in C:\Users\Dan\AppData\Roaming\Python\Python37\site-packages as suggested
I can see this module in tools-->plugins but not in tools -->modules
When i import MircoWebsrv2 i get : Import Error: no module named 'microwebsrv2'
Do you know how to solve that ?
Kindly
Daniel

aivarannamaa, 2021-01-25:
@dnguyen76, if I'm not mistaken then you need this library on your device, not on your local disk.

Normally I would say that you need to use "Tools => Manage packages", but it looks like this library is not on PyPI, so I guess you need to download the zip and upload it manually.

aivarannamaa, 2021-01-25:
You want to use it with a MicroPython device, right?

dnguyen76, 2021-01-25:
Yes with Micropython for ESP32 
More information, I succed to import ssd1306 module and so see it in module list, but i can't with MicroWebServ2)

aivarannamaa, 2021-01-25:
I haven't used this library myself, but, as I said, it looks like you have to download the zip, extract it to your local drive and upload relevant parts from it into your device's lib directory (in Thonny "View => Files", go your extraction directory in the upper pane and lib directory in the lower pane, select files, right click, select "Upload to /lib")

dnguyen76, 2021-01-25:
OK thanks very much
It's works

aivarannamaa, 2021-05-18:
Will be fixed in Thonny 3.3.8

closed 2021-05-18 by aivarannamaa